stlink-org/stlink #672: st-util cant find any st-link/V2 devices

mahershaikh1, 2018-02-21:
Hi,

I am very new to a lot of what I have been working on so I will try to describe my problem as carefully as possible. I am using an STM32F411E-DISCO on a Mid 2012 Macbook pro with version 10.13.3 I just used homebrew with the command "brew install stlink" to get stlink. 

When entering "st-util" into terminal, I get the error:
"WARN usb.c: Couldn't find any ST-Link/V2 devices"

Any ideas whats wrong?

Thanks

xor-gate, 2018-02-21:
I must say I have not tested on latest macOS 10.13 myself. It did work before. Maybe there is some permission problem you could try to run it as root with:  `sudo st-util --probe`.

xor-gate, 2018-02-21:
```
jerry@Jerrys-iMac:[~]
 > sw_vers
ProductName:	Mac OS X
ProductVersion:	10.13.3
BuildVersion:	17D47
jerry@Jerrys-iMac:[~]
 > st-info --probe
Found 1 stlink programmers
 serial: 503f7206506752553329033f
openocd: "\x50\x3f\x72\x06\x50\x67\x52\x55\x33\x29\x03\x3f"
  flash: 262144 (pagesize: 256)
   sram: 32768
 chipid: 0x0427
  descr: L1 Medium-Plus-density device
```

maxmunzel, 2018-11-04:
For me the homebrew install works just fine with macOS Mojave on a late 2017 MBP and a NUCLEO-F401RE board (also a F4/V2).

```bash
$st-util
st-util 1.5.1
2018-11-04T16:46:33 INFO common.c: Loading device parameters....
2018-11-04T16:46:33 INFO common.c: Device connected is: F4 device (Dynamic Efficency), id 0x10016433
2018-11-04T16:46:33 INFO common.c: SRAM size: 0x18000 bytes (96 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 16384 bytes
2018-11-04T16:46:33 INFO gdb-server.c: Chip ID is 00000433, Core ID is  2ba01477.
2018-11-04T16:46:33 INFO gdb-server.c: Listening at *:4242...
``` 
[image]

xor-gate, 2018-12-28:
There is no such thing as a "normal serial port" on the programmers except for the v2-1 on some nucleo boards which have USB-serial. `st-info --probe` displays only output when a microcontroller is detected.

Reinual, 2019-01-02:
Thx,  I do see my St-Link V2 debugger using command:

st-info --probe

However, I don't see the information on the connected bluepill device:

Found 1 stlink programmers
 serial: 533f7006483f49505128173f
openocd: "\x53\x3f\x70\x06\x48\x3f\x49\x50\x51\x28\x17\x3f"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device

Any clue why that would be?

pan-mroku, 2019-03-15:
@Reinual I think your problem is different than OP. probe has found your programmer but can't get info (flash: 0, sram: 0, chipid: 0). I am having the same problem with some faulty program that uses DMA and I2C (haven't found which one is the culprit yet). In order to unlock it I use https://github.com/texane/stlink/issues/705#issuecomment-395981733:
1. Press reset button on the board and hold it
2. openocd -f interface/stlink-v2-1.cfg  -f target/stm32f4x.cfg -c "init; reset halt; stm32f4x unlock 0; reset halt; exit"
3. When opencd starts waiting for target to halt release the button (you will notice a timeout when it waits)
4. st-flash erase
  It will fail for the first time
5. st-flash erase
  this will erase the program in flash and everything should get back to normal. As long as you don't program the same binary again of course. ;)

Nightwalker-87, 2020-03-20:
@mahershaikh1: Have you found a solution for this problem? If yes, please let us know.

Nightwalker-87, 2020-04-06:
As shown by **xor-gate** and **maxmunzel** the original problem does obviously not appear to be related to the `stlink` toolset itself - one feedback addresses the same macOS version and the other a similar STM32F4 device. Thus it seems appropriate to close this issue as resolved now.

closed 2020-04-06 by Nightwalker-87