stlink-org/stlink #673: Support for STM32F411CEU6

Hoel, 2018-02-21:
Apparently the STM32F411 is not supported:

```
2018-02-21T19:47:37 INFO common.c: Loading device parameters....
2018-02-21T19:47:37 WARN common.c: unknown chip id! 0x20002e5c
```
Is  it possible to add support for F411?
Thanks

xor-gate, 2018-02-21:
I think it should be fairly easy to add, not sure if the flash loader is STLINK_FLASH_TYPE_F4:

* https://github.com/texane/stlink/blob/master/include/stlink/chipid.h
* https://github.com/texane/stlink/blob/master/src/chipid.c

Hoel, 2018-02-21:
ha, thank you i didnt realized it was just a matter of adding definition here, alright i'll give a try and make a PR if everything is fine.

xor-gate, 2018-02-21:
You should double check the chipid is correct (from datasheet), maybe the wrong register is read.

Hoel, 2018-02-21:
OK thanks, according RM0383 chipid is  0x431, same as STLINK_CHIPID_STM32_F411RE

xor-gate, 2018-02-22:
There is something wrong as your issue reads `0x20002e5c` where the last three nibbles should be the chipid. Which is `0xe5c` and not the same as `0x431`, maybe the wrong register is read by stlink tools.

daniel-santos, 2019-09-05:
I'm getting:

```
$ st-flash read out.bin 0x20000000 4096
st-flash 1.5.1
2019-09-05T05:10:03 INFO common.c: Loading device parameters....
2019-09-05T05:10:03 WARN common.c: unknown chip id! 0x20016469
```
with an stlink v2 hooked up to a nucleo-g474re. Of course, this is a hack its self because I used a windows VM to start the on-board stlink v3 in "firmware mode" for updating the firmware and left it there so it would go high-z on its debug pins so I could use a v2 debugger via the swd connector on the board.  All of this worked, but now the software doesn't like the target chip. :(

This is an stm32g474ret6u

Nightwalker-87, 2020-04-02:
@Hoel: As mentioned before, it is obvious that the STM32F411CEU6 device reports the false chip-ID to the stlink tools. I would presume that this chip is defective.
@daniel-santos: Your STM32G474RET6U reports the correct chip-ID `0x469`. Please note that support for this device was not present until #822, thus not available in Release v1.6.0 either. It will be supported in the upcoming Release v1.6.1. Please use the `development` branch until then.

closed 2020-04-02 by Nightwalker-87