stlink-org/stlink #674: Getting printf output on command line from STM32F411E-Discovery

Matombo, 2018-02-26:
- [X] Programmer/board type: Stlink/v2-onboard
- [ ] Programmer firmware version: ??? (don't know how to find out)
- [X] Operating system: Linux 4.15
- [X] Stlink tools version and/or git commit hash: 1.5.0
- [X] Target chip (and optional board): STM32F411E-Discovery

I have a noob question:
I try to get printf output from my board without using an serial adapter, i already read that st-term got ditched in favor of ARM semihosting but every explanation on how to use it, i can find, involves some kind of ide.
And since i'm using a makefile project and st-flash directly that doesn't help much. So i'm currently completly clueless und which tool i should use or google to get some output.

xor-gate, 2018-02-26:
You need to link a special rdimon library (glibc) into your application, or write a little assembly code.

http://bgamari.github.io/posts/2014-10-31-semihosting.html
http://forum.segger.com/index.php?page=Thread&threadID=1762
https://github.com/xor-gate/arm-by-example/blob/master/stm32l152rct-discovery/lesson-4/main.s

Good luck.

closed 2018-02-26 by xor-gate

Matombo, 2018-03-01:
So i figured it out how to get console output using semihosting:

- My starting point is a Makefile project created by CubeMX
- In the Makefile remove "-lnosys" from LIBS variable and replace "-specs=nano.specs" with "-specs=rdimon.specs" in LDFLAGS variable (this makes the compiler include the semihosting librarys)
- Use stdio.h and printf in your code (note that if you included the semihosting libraries and are making use of them your program will freeze if the board is not connected to a debugging programm reciving the output)
- Download and install OpenOCD (which as it looks like can actually be used as a full replacement for stlink)
- Create a config file (filename.cfg):
```
source [find interface/stlink-v2.cfg]

source [find target/stm32f4x.cfg]

init

arm semihosting enable

reset
```
- let OpenOCD connect to your board with "openocd -f filename.cfg" (note that the reset command in the config makes the programm run from start. The reset must be issued there for the "arm semihosting enable" option to take effect)
- ???
- PROFIT!!! (you should get the printf messages)

- ps you can use gcc-arm-eabi to connect to localhost:3333 for gdb debugging
- pps to issue openocd commands from gdb prefic them with "monitor" (e.g. "monitor reset halt" and "monitor reset run")
- ppps I'm not 100% sure on my explanations here, it's just what i think makes sense

xor-gate, 2018-03-01:
Thank you for your feedback, but this is related to usage of OpenOCD and not st-util. You should run `st-util --semihosting` option.

Matombo, 2018-03-02:
Thanks for the info, the manpage of st-util was missing the "--semihosting" flag, but i see, you already fixed that ^^.

I have another question now: If you run "st-util --semihosting" the execution is paused untill you connect to it with gdb. Is there a way to directly start the execution if you just want to get some output (like with openocd where you can choose between "reset halt" and "reset run" in your config files)

xor-gate, 2018-03-02:
Semihosting uses the `bkpt` assembly instruction which hits a breakpoint. Only the debugger is able to get the microcontroller out of this.

Wfuger, 2020-03-31:
@Matombo thank you that is incredibly helpful!

commit 1f18a18a (2018-03-01): doc/man: Add semihosting parameter documentation, fix styling issue with double dashhes. Fixes #674.