stlink-org/stlink #677: not work mac os x binary on snow leopad

yamori813, 2018-03-02:
I try to stlink-1.3.0-macosx-amd64 binary on snow leopad(10.6.8) but now work.

bash-3.2$ ./st-flash 
Illegal instruction

I seem newer Xcode make not compatible Mach-O binary for snow leopad.

Regards

xor-gate, 2018-03-02:
On such an old Mac OS X version you have to compile your own version https://github.com/texane/stlink/blob/master/doc/compiling.md. Probably you first need to compile or get a binary version of libusb as that is the only dependency. Sorry!

yamori813, 2018-03-02:
thanks for your comment. current code can't compile on gcc4. Because of for(int i;;) is error on gcc4.

xor-gate, 2018-03-02:
variable declaration inside loops is supported since C99, not sure if GCC 4 supports it. you would be better of to get yourself a raspberry pi and run stlink on there.

yamori813, 2018-03-02:
I make pr. https://github.com/texane/stlink/pull/678

closed 2020-03-18 by Nightwalker-87