stlink-org/stlink #679: Cannot connect to STM32VL under Linux

cedricboudinet, 2018-03-06:
Hello

I am trying to connect to a STM32VL discovery board under Debian 9 32bits.
Unfortunately, st-info seems to be unable to detect the board:

$ st-info --probe
Found 0 stlink programmers

But the board is connected:
$ lsusb
Bus 002 Device 004: ID 0483:3744 STMicroelectronics STLINK Pseudo disk

I checked st-info with a STM32F407 discovery board (STLINK/V2) and it works.

Does it mean that stlink for STLINK/V1 is not working under Linux or am I doing something wrong?

Regards
Cedric

xor-gate, 2018-03-06:
I think the stlink v1 programmers are claimed by the kernel and for normal users it is with the default permissions possible to detach the kernel driver for the emulated disk drive and send raw programming commands. Not much people use the stlinkv1 but this is what I found: https://github.com/texane/stlink/blob/ba898b1c9c75800b566e8b1983ff7643e104a994/src/sg.c#L62-L74

cedricboudinet, 2018-03-12:
/etc/modprobe.d/stlink_v1.conf has been automatically installed and seems to be taken into account.
When I plug the card, dmesg gives:
[ 5290.649419] usb 2-1.2: USB disconnect, device number 5
[ 5293.944538] usb 2-1.2: new full-speed USB device number 6 using ehci-pci
[ 5294.061008] usb 2-1.2: New USB device found, idVendor=0483, idProduct=3744
[ 5294.061021] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5294.061027] usb 2-1.2: Product: STM32 STLink
[ 5294.061033] usb 2-1.2: Manufacturer: STMicroelectronics
[ 5294.061038] usb 2-1.2: SerialNumber: PÿnPgRUG\xffffffc2\xffffff87
[ 5294.061774] usb-storage 2-1.2:1.0: USB Mass Storage device detected
[ 5294.061958] usb-storage 2-1.2:1.0: device ignored

I think that the last line means that the device is not mounted as a mass storage.

I just tried something else: I connected the board and launched st-util: it manages to connect.
It seems the issue is only with st-info.

cedricboudinet, 2018-03-12:
STLINK/V1 is ignored in stlink_probe_usb_devs (https://github.com/texane/stlink/blob/b69c1d51b03a4345c4d478ac1d8c2088002cd36a/src/usb.c#L976 and https://github.com/texane/stlink/blob/b69c1d51b03a4345c4d478ac1d8c2088002cd36a/src/usb.c#L1000).
I changed the test to make it took into account but then the stlink_print_info output seems to be broken (e.g. for openocd and serial). Should the display be adapted to the stlink version ?

Nightwalker-87, 2020-03-21:
@cedricboudinet: Which release version did you use?

Nightwalker-87, 2020-04-09:
This was reported using Release **v1.5.0**. As this ticket has not seen any further activity since, we need someone with a ST-Link-v1 programmer (PID: 3744, `STLINK_USB_PID_STLINK`) to test  this again with `st-info`.
@cedricboudinet: I find that you keep maintaining a fork of this project. Is the related hardware still present and are you able to test?

cedricboudinet, 2020-04-09:
@Nightwalker-87 I am not able right now be because of the Covid. The card is in my office but it is still working.

grevaillot, 2020-04-09:
I've got an stm32f100 discovery somewhere, I think it match that descrption, i'll check

Nightwalker-87, 2020-04-09:
@cedricboudinet: That does not matter, we have time. Good to hear though the hardware is still available.
@grevaillot: 8-) :+1:

grevaillot, 2020-04-09:
ok, i just tested a clean develop branch build :+1: 

>  % lsusb|grep ST
> Bus 002 Device 035: ID 0483:3744 STMicroelectronics ST-LINK/V1

erase works, 
>  % ./build/Release/bin/st-flash erase 
> st-flash 1.6.0-145-g38c8b8b
> 2020-04-09T14:52:11 INFO common.c: F1xx Value Line: 8 KiB SRAM, 128 KiB flash in 1 KiB pages.
> Mass erasing

full flash write is ok

>  % ./build/Release/bin/st-flash write /tmp/saved.bin 0x8000000
> st-flash 1.6.0-145-g38c8b8b
> 2020-04-09T14:56:17 INFO common.c: F1xx Value Line: 8 KiB SRAM, 128 KiB flash in 1 KiB pages.
> 2020-04-09T14:56:17 INFO common.c: Attempting to write 131072 (0x20000) bytes to stm32 address: 134217728 (0x8000000)
> Flash page at addr: 0x0801fc00 erased
> 2020-04-09T14:56:20 INFO common.c: Finished erasing 128 pages of 1024 (0x400) bytes
> 2020-04-09T14:56:20 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
> 2020-04-09T14:56:20 INFO flash_loader.c: Successfully loaded flash loader in sram
> 128/128 pages written
> 2020-04-09T14:56:28 INFO common.c: Starting verification of write complete
> 2020-04-09T14:56:31 INFO common.c: Flash written and verified! jolly good!
> 

but probe fails ..
>  % ./build/Release/bin/st-info --probe
> Found 0 stlink programmers
>

grevaillot, 2020-04-09:
Interesting, having a second stlink connected makes it probe the v1: 

>  % ./build/Release/bin/st-info --probe
> Found 1 stlink programmers
>  serial:     
>  hla-serial: ""
>  flash:      131072 (pagesize: 1024)
>  sram:       8192
>  chipid:     0x0420
>  descr:      F1xx Value Line
> 

the stlink_probe() is filtering out the stlink v1 usb pid (pid is missing). But then, the usb open stuff takes care of the v1 first, and probe succeed. That part of code need some care regarding the usb pid checks, this does not really look aligned.

Nightwalker-87, 2020-04-09:
Does not look good to me either, it should be working in any combination and method applied. This is a bug.

closed 2020-04-10 by Nightwalker-87