stlink-org/stlink #681: Attempting to flash to STM32L051K8 corrupts flash memory

WRansohoff, 2018-03-09:
- [X] Programmer/board type: Stlink/v2
- [X] Programmer firmware version: Unknown; generic St-link module.
- [X] Operating system: Linux (Ubuntu 16.04)
- [X] Stlink tools version and/or git commit hash: v1.5.0
- [X] Stlink commandline tool name: st-util
- [X] Target chip (and optional board): STM32L051K8

Attempting to flash a .elf to the chip results in a series of errors indicating an STM32L1 loader is being used (quoted further down), and subsequent connections fail with the following strange error message:

```
2018-03-08T23:04:41 WARN common.c: Invalid flash type, please check device declaration
2018-03-08T23:04:41 INFO gdb-server.c: Chip ID is 00000000, Core ID is  0bc11477.
```

That state persists until the chip is 'fixed' by connecting to it during a hardware reset via ST's ST-Link Utility on a Windows machine. Here is the output that st-util gives during the problematic flashing operation:

```
st-util 1.5.0
2018-03-08T23:04:03 INFO usb.c: -- exit_dfu_mode
2018-03-08T23:04:03 INFO common.c: Loading device parameters....
2018-03-08T23:04:03 INFO common.c: Device connected is: L0x3 device, id 0x10386417
2018-03-08T23:04:03 INFO common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 128 bytes
2018-03-08T23:04:03 INFO gdb-server.c: Chip ID is 00000417, Core ID is  0bc11477.
2018-03-08T23:04:03 INFO gdb-server.c: Listening at *:4242...
2018-03-08T23:04:13 INFO gdb-server.c: Found 4 hw breakpoint registers
2018-03-08T23:04:13 INFO gdb-server.c: GDB connected.
2018-03-08T23:04:14 INFO common.c: Attempting to write 128 (0x80) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2018-03-08T23:04:14 INFO common.c: Finished erasing 1 pages of 128 (0x80) bytes
2018-03-08T23:04:14 INFO common.c: Starting Half page flash write for STM32L core id
2018-03-08T23:04:14 INFO flash_loader.c: Successfully loaded flash loader in sram
2018-03-08T23:04:17 ERROR flash_loader.c: flash loader run error
2018-03-08T23:04:17 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2018-03-08T23:04:17 WARN common.c: 
write_half_pages failed == -1
  0/  1 pages written
2018-03-08T23:04:17 INFO common.c: Starting verification of write complete
2018-03-08T23:04:17 INFO common.c: Flash written and verified! jolly good!
2018-03-08T23:04:17 INFO common.c: Attempting to write 128 (0x80) bytes to stm32 address: 134217856 (0x8000080)
Flash page at addr: 0x08000080 erased
2018-03-08T23:04:17 INFO common.c: Finished erasing 1 pages of 128 (0x80) bytes
2018-03-08T23:04:17 INFO common.c: Starting Half page flash write for STM32L core id
2018-03-08T23:04:17 INFO flash_loader.c: Successfully loaded flash loader in sram
2018-03-08T23:04:20 ERROR flash_loader.c: flash loader run error
2018-03-08T23:04:20 WARN common.c: l1_stlink_flash_loader_run(0x8000080) failed! == -1
2018-03-08T23:04:20 WARN common.c: 
write_half_pages failed == -1
  0/  1 pages written
2018-03-08T23:04:21 INFO common.c: Starting verification of write complete
2018-03-08T23:04:21 INFO common.c: Flash written and verified! jolly good!
[ ... repeats a few times ... ]
```

Expected/description:
The program should flash susccessfully. But despite reporting success, when I try to step through it in GDB, the chip jumps directly from the reset location to 0x20000010. So I gotta figure that something went wrong somewhere.

xor-gate, 2018-03-09:
Thats a weird problem, I could not immediate tell what causes the problem. I think the L1 loader is compatible with the L0 chips as they are almost the same.

WRansohoff, 2018-03-09:
It does seem a little strange - let me know if there are any debug flags or something that I should turn on to provide more information.

cstratton, 2018-04-20:
Try flashing a flat binary with st-flash rather than an elf.

Try reading back the flash to a file and comparing, or examining it with gdb, or example
x/32x 0x8000000

You should see the start of your vector table...
Where did your binary come from, and how do you know it is good?

Also given that your address seems to be in RAM, what is the state of your BOOT pin?

WRansohoff, 2018-04-23:
I'll take a look - I've been meaning to learn how to flash .bin files instead of uploading .elf's with a debugger, anyways.

The code was one of ST's examples, but I'll take another look at where the vector table is located.

The BOOT0 pin is pulled to ground, so I don't think it should be booting into RAM.

I'll get back on those questions within a day or two - thank you for the advice!

WRansohoff, 2018-04-24:
Alright, I tried flashing the L0 test program (specifically, ST's LSI initialization example) to an L031K6 Nucleo-32 board, and that works fine.

On the L051K8, however, I still observe the issue (after adjusting the project to use the right device header files and compiler options). The main difference is that I am using a generic USB dongle to program the L051K8 board, and the onboard ST-Link/V2-1 debugger for the L031K6 Nucleo board. I tried updating the firmware on the USB debugger, but that didn't help.

I verified that the vector table is correctly placed with `arm-eabi-none-nm` in both cases: `08000000 R g_pfnVectors`

Flashing a .bin file to the L051K8 with `st-flash write main.bin 0x08000000` throws an error but still reports success. Connecting to the chip afterwards shows that the program is stuck at 0xFFFFFFFE, though.

Here is the output from `st-flash`:

`
st-flash 1.5.0
2018-04-23T23:26:42 INFO common.c: Loading device parameters....
2018-04-23T23:26:42 INFO common.c: Device connected is: L0x3 device, id 0x10386417
2018-04-23T23:26:42 INFO common.c: SRAM size: 0x2000 bytes (8 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 128 bytes
2018-04-23T23:26:42 INFO common.c: Ignoring 4 bytes of 0x00 at end of file
2018-04-23T23:26:42 INFO common.c: Attempting to write 18404 (0x47e4) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08004780 erased
2018-04-23T23:26:43 INFO common.c: Finished erasing 144 pages of 128 (0x80) bytes
2018-04-23T23:26:43 INFO common.c: Starting Half page flash write for STM32L core id
2018-04-23T23:26:43 INFO flash_loader.c: Successfully loaded flash loader in sram
2018-04-23T23:26:46 ERROR flash_loader.c: flash loader run error
2018-04-23T23:26:46 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2018-04-23T23:26:46 WARN common.c: 
write_half_pages failed == -1
142/143 pages written
2018-04-23T23:27:03 INFO common.c: Starting verification of write complete
2018-04-23T23:27:04 INFO common.c: Flash written and verified! jolly good!
`

But looking at the Flash memory with ST's GUI utility, I do see the .bin file's hex starting at 0x08000000; it looks like the vector table at the start of the program. And the written memory looks like it ends at 0x080047E0, which seems right for the size of the program.

I dunno - if this isn't something that anyone else can replicate, I suppose that it could be the chips or boards that I'm using.

bourou01, 2018-05-04:
Hi,

Same issue with blue-pill board (STM32F103C8X)
After every program with 'arm-none-eabi-gdb', I got successful, the program run well in the board but when I try to program it again I got theses issues:

st-util
-------
[abdullatif@Host-001 ~]$ st-util
st-util 1.5.0
2018-05-04T23:47:27 INFO common.c: Loading device parameters....
2018-05-04T23:47:27 WARN common.c: Invalid flash type, please check device declaration
2018-05-04T23:47:27 INFO gdb-server.c: Chip ID is 00000000, Core ID is  00000000.
2018-05-04T23:47:27 INFO gdb-server.c: Listening at *:4242...
2018-05-04T23:47:40 INFO gdb-server.c: Found 0 hw breakpoint registers
2018-05-04T23:47:40 INFO gdb-server.c: GDB connected.

gdb
-----
[abdullatif@Host-001 build]$ arm-none-eabi-gdb Stm32blink.elf
GNU gdb (GDB) 8.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from Stm32blink.elf...done.
(gdb) target extended-remote :4242
Remote debugging using :4242
0x00000000 in ?? ()
(gdb) load
Loading section .isr_vector, size 0x10c lma 0x8000000
Load failed
(gdb) 

I have to flash the bootloader with 'st-flash' if I what the board to work again.

WRansohoff, 2018-05-12:
I'm not convinced that that F103 issue is the same as what I've been seeing.

The original title of this issue was incorrect, sorry; the behavior occurs on the following two devices:

* STM32L052K8

* STM32L082KZ

I looked at `src/chipid.c` and saw that only the reference manuals for L0x1 and L0x3 chips were referenced for the settings; could L0x2 chips have any major differences?

jakeypoo, 2019-02-28:
Having Similar issues with an STM32L151CC. 
Everything programs fine, just throws the same error and fails on verification step.

```
st-flash --reset write 'build/bin'/uartbridge-mbiot.bin 0x08000000
st-flash 1.4.0-57-g7651d21
2019-02-28T09:33:08 INFO common.c: Loading device parameters....
2019-02-28T09:33:08 INFO common.c: Device connected is: L1 Medium-Plus-density device, id 0x10f86427
2019-02-28T09:33:08 INFO common.c: SRAM size: 0x8000 bytes (32 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 256 bytes
2019-02-28T09:33:08 INFO common.c: Attempting to write 14376 (0x3828) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08003800 erased
2019-02-28T09:33:08 INFO common.c: Finished erasing 57 pages of 256 (0x100) bytes
2019-02-28T09:33:08 INFO common.c: Starting Half page flash write for STM32L core id
2019-02-28T09:33:08 INFO flash_loader.c: Successfully loaded flash loader in sram
2019-02-28T09:33:12 ERROR flash_loader.c: flash loader run error
2019-02-28T09:33:12 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2019-02-28T09:33:12 WARN common.c: 
write_half_pages failed == -1
 55/ 56 pages written
2019-02-28T09:33:25 INFO common.c: Starting verification of write complete
2019-02-28T09:33:25 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```

Nightwalker-87, 2020-03-19:
@WRansohoff: Please verify if the problem still exists in Release v1.5.1.

WRansohoff, 2020-03-24:
Sure, happy to check whether this still happens with STM32L0 category-2 chips.

Most of my boards are in storage atm though, so I probably won't be able to do it until I finish moving in May. Sorry about that.

Nightwalker-87, 2020-06-03:
ping @WRansohoff

Nightwalker-87, 2020-07-19:
@WRansohoff: Any update on this?

sansil, 2020-07-21:
Hi, i have the same issue with STM32L433RC.
I'm using st-util 1.6.1

lapers, 2020-07-31:
Try to erase flash of your MCU
Use this article: https://electronics.stackexchange.com/questions/204996/stm32-st-link-cannot-connect-to-mcu-after-successful-programming
 
Maybe you forgot to mark SWD /JTAG pins at STM32Cube

Ant-ON, 2021-03-26:
@Nightwalker-87 Sorry, I have no boards with L series chip. The problem here is strange, a quick look at the code and reading the documentation did not give any results.

Nightwalker-87, 2021-03-26:
Anyone else? Indeed there are still a few L-related other tickets as well, so folks with such boards would clearly help.

V1DEOJAMES, 2021-04-13:
I'm also experiencing an issue when trying to flash many boards using st-flash 1.6.1.

Seems to default to very slow page write.
ERROR flash_loader.c: flash loader run error
WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
WARN common.c: 
write_half_pages failed == -1
452/453 pages written

Ant-ON, 2021-04-13:
@V1DEOJAMES Can you try https://github.com/Ant-ON/stlink/tree/fl_fix branch?

Nightwalker-87, 2021-05-14:
Ping @V1DEOJAMES

V1DEOJAMES, 2021-05-14:
Apologies, my issue ended up being unrelated and was solved by using a different programming board (previous one was damaged by a bad USB hub). I did test the branch at the time but it made no difference for me.

Nightwalker-87, 2021-05-15:
Ok, so your observations were obviously not related to this issue. Thx for the feedback. :+1:

Nightwalker-87, 2021-05-26:
@wolsty7 Can you try to reproduce this issue with your hardware using older versions of the toolset if you find some time? We are still trying to figure out if this issue has been fixed already and where... As far as I'm concerned we were limited on hardware from the L-series needed for testing. Any help would be greatly appreciated.

Nightwalker-87, 2021-08-14:
@gszy Can you help us with this one?

gszy, 2021-08-14:
I’m sorry, I don’t have any L0s to test.

Nightwalker-87, 2021-08-14:
I see. However thanks for the feedback. Feel free to take a look at other issues if you fancy to to so. :+1:

Nightwalker-87, 2022-01-06:
I have spotted a false `bootrom_base` hex address for the L0 series.
According to the viewed reference manuals the is a missing digit: `0x1ff0000` --> `0x1ff00000`
This appears to affect other L0-chips as well. Please check `chipid.c`.

logibburrell, 2022-05-07:
DEVICE: NUCLEO-H7A3ZI-Q

_Reproduction_
st-util -v99
```
st-util 1.7.0-186-gc4762e6-dirty
Failed to parse flash type or unrecognized flash type
2022-05-06T21:34:59 DEBUG common.c: *** looking up stlink version
2022-05-06T21:34:59 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2022-05-06T21:34:59 DEBUG common.c: stlink pid     = 0x374e
2022-05-06T21:34:59 DEBUG common.c: stlink version = 0x3
2022-05-06T21:34:59 DEBUG common.c: jtag version   = 0x9
2022-05-06T21:34:59 DEBUG common.c: swim version   = 0x0
2022-05-06T21:34:59 DEBUG common.c: stlink current mode: mass

```

_Resolution_
Running "st-flash --connect-under-reset erase 0x08000000 0x80" and then and restarting device.
Note: uploading new elf with --connect-under-reset did not help. 

The culprit that appeared to cause this was the "Examples/FLASH/FLASH_EraseProgram" project in the nucleo repo area.

Nightwalker-87, 2022-05-07:
> I have spotted a false `bootrom_base` hex address for the L0 series. According to the viewed reference manuals the is a missing digit: `0x1ff0000` --> `0x1ff00000` This appears to affect other L0-chips as well. Please check `chipid.c`.

In order to finally resolve this issue in this context it would be great to verify this with the L052K8 however.
Well possible that there were several reasons for this to appear.

BRA1L0R, 2022-10-17:
I am having this same issue on an STM32L072CZ flashed through another nucleo's ST-Link.

When I try to start `st-util --no-reset` I get:
```
Unsupported Target (Chip ID is 0000000000, Core ID is 0x0bc11477).
```

But when I try the same command without `--no-reset` it works just fine.

Also, when I try loading an elf `cortex-debug` (a vscode extension) I get this error:
```
ERROR flash_loader.c: flash loader run error
WARN common.c: l1_stlink_flash_loader_run(0x8000080) failed! == -1
```

Nightwalker-87, 2022-12-29:
@BRA1L0R Which release version or commit did you use for testing? Can you `git bisect `up to the current HEAD of the `develop` branch to give us an update?

Nightwalker-87, 2023-03-19:
@WRansohoff We are about to finally resolve this issue. Are you still available for testing?

Nightwalker-87, 2023-03-19:
@BRA1L0R Your problem is documented in #1203. As far as i can tell it's not related to this issue.

Nightwalker-87, 2023-04-30:
@chenguokai Hi there. I've reviewed this issue and found that the observed behaviour is related to the flash-loader implementation you contributed in 2020 (which I believe is a very valuable contribution to this project).
Would you be interested to look into this topic and some obviously related issues?
I'd also appreciate to get in touch with you to find out more regarding the technical background of your implementation.

Nightwalker-87, 2023-05-10:
We need contributors here with L0 hardware for testing purposes.

Nightwalker-87, 2023-05-20:
@mynameisdaniil I think you can help us here as this thread appears related to the problem you faced in #1225.

nemtsov, 2023-05-30:
I'm seeing possibly the same issue on my `STM32L010F4` MCU. I'm debugging it through a `NUCLEO-L010RB` (with the STLINK programmer broken off) on a mac `13.3.1`, with `stlink` `v1.7.0` installed via Homebrew. 

Here's a [minimal repro in a gist](https://gist.github.com/nemtsov/513ed7e110b6299e8cf64355e4ac40e8) along with the terminal output. When the final `MOV` is commented out everything works and I get the `Flash written and verified! jolly good!` in the end of the `st-flash --format ihex write minimal.hex` command. If the 28th `MOV` is uncommented, I can reliably get the `Flash loader run error`. 

This happens in all with the `BOOT0` pin in both 0 and 1 configuration (since someone asked above). And, importantly, if I take the failing `minimal.hex` and use the `STM32CubeProgrammer` to flash it there are no issues. Happy to try things out on this hardware.

Nightwalker-87, 2023-05-30:
@nemtsov Thanks for the helpful feedback. I'll look into this in more detail during the following weekend.
It may be a good advice to locate the lines of interest in the source code, starting from the received output message in order to trace it back to the root.

nemtsov, 2023-05-30:
@Nightwalker-87 one other thing I should note (in case this helps) is that I can repro this with different source code. I started that gist with the [actual source I'm writing](https://gist.github.com/nemtsov/513ed7e110b6299e8cf64355e4ac40e8/revisions#diff-d284c3c456415d7c778054b688e0e4b8dabf2cab9ad2244f9f5a53abe9e393bb) and then slowly changed all instructions to `MOV` to make the repro source cleaner. 

I'm not sure if that's what's happening (or the bug just gives this appearance), but what it feels like is that it isn't so much which instructions that are used but how many of them. It's as if there's a limit on the space for instructions and I can either fill it with 27 `MOV`s or 15 `LDR, MOV, STR, ORR, etc.`, and adding one more puts it over the edge. Or at least experimentally that's how I'm triggering this issue.

Nightwalker-87, 2023-05-30:
@nemtsov This assembler based approach is very interesting (i've just started to learn avr-asm). However we still need the look at the current implementation in C as well (may it be implemented well or not) at least to decide whether it is worth fixing or if an alternative approach is favourable. I currently have no idea on what we will find in this part of the code. :confused:

Mk-arc, 2023-06-06:
It seems I have the same problem. Tested with STM32L063R8Tx on a custom made board.
So far I have been unable to program this board with st-link.

Flashing works fine with STM32CubeProgrammer and the same st-link probe (V2) or with other probes, e.g. J-Link.

If there is anything I should test, let me know.

Nightwalker-87, 2023-06-06:
Hi everybody. I finally found some time to look into this again.

This is likely to (also) be a timing-related problem. Check `flash_loader.c` L357ff:

```
/* 
 * This piece of code used to try to spin for .1 second by waiting doing 10000 rounds of 10 µs.
 * But because this usually runs on Unix-like OSes, the 10 µs get rounded up to the "tick"
 * (actually almost two ticks) of the system. 1 ms. Thus, the ten thousand attempts, when
 * "something goes wrong" that requires the error message "flash loader run error" would wait
 * for something like 20 seconds before coming up with the error.
 * By increasing the sleep-per-round to the same order-of-magnitude as the tick-rounding that
 * the OS uses, the wait until the error message is reduced to the same order of magnitude
 * as what was intended. -- REW.
 */

    // wait until done (reaches breakpoint)
    timeout = time_ms() + 500;
    while (time_ms() < timeout) {
        usleep(10000);

        if (stlink_is_core_halted(sl)) {
            timeout = 0;
            break;
        }
    }

    if (timeout) {
        ELOG("Flash loader run error\n");
        goto error;
    }
```

To me this does not look like a well considered/written implementation.
Does anybody have ideas on how to rethink/rewrite/improve the function `stlink_flash_loader_run( )` ?

Please also take note of the remaining code parts from the old source file `flashloader.c` from L418ff.
Up to now these have not been removed along with newer implementations in order to preserve functionality (which is not always known and/or documented).

Nightwalker-87, 2023-06-17:
@bourou01 / st-util 1.5.0 / Device: STM32F103C8X "Bluepill"
--> unrelated issue, board is not detected, see `INFO gdb-server.c: Chip ID is 00000000, Core ID is 00000000.`
Likely related to faulty hardware or faked chip; refer to section "NOTE: Chinese Fake-Chips CKS32F103C8T6 or CS32F103C8T6 being marked as "STM32F103C8T6"" in `/doc/tutorial.md` for more details.

Nightwalker-87, 2023-06-17:
Time for a summary in order to draw this to a close:

1) @WRansohoff / Release st-flash 1.5.0 / Devices: STM32L052K8 & STM32L082KZ
2) @jakeypoo / st-flash 1.4.0-57-g7651d21 / Device: STM32L151CC
3) @sansil / st-util 1.6.1 / Device: STM32L433RC
4) @Mk-arc / No report on used version/commit! / Device: STM32L063R8Tx

Nightwalker-87 on on Jan 6, 2022:
--> false `bootrom_base` hex address for the L0 series.
According to the viewed reference manuals the is a missing digit: `0x1ff0000` --> `0x1ff00000`
This has been fixed in commit e62b9e1f2a1b6dea7e9ccb053c14c2c6ad0b57d7.

Nightwalker-87, 2023-06-18:
Note, that this ticket serves as the main thread for the equal or similar issues: #1203, #1225, #1253, #1289

@Ant-ON Here we have @nemtsov , @Mk-arc and @mynameisdaniil with suitable hardware for testing.

Here is how I'd proceed:

1) Firstly, I recommend volunteers to do some testing with with their hardware based on the latest `testing` branch (commit: b72f5b5acf8997a299d9eac4d6cc2dfc5322e144) to give an overview of the current state. Ideally this should be done by all involved contributors listed above.

2) Referring to your thoughts in https://github.com/stlink-org/stlink/issues/1225#issuecomment-1596079595 we should then make use of option **b)**. This would give additional info on the resulting effects.

3) Based on the results we shall decide on if to disable half-page-writing for L0/L1 as a interim solution.
If this should turn out to be the case, then we'll close this topic for now, and open a new issue in order to to fix the implementation for half-page-writing as a feature-request.

Ant-ON, 2023-06-18:
@Nightwalker-87
RM0038 Rev 18 at pg. 65 has warning for Half Page Write:
> Warning: Half Page Write is possible only from SRAM, from external memory or from bank 1 to bank 2 and vice versa.

RM0367 Rev 8 at pg. 82 has me a conspicuous warning:
> This protocol can be used either through application code running from RAM or through DMA with application code running from RAM or core sleeping.

Therefore, I believe that the code can be disable.

Nightwalker-87, 2023-06-18:
@Ant-ON Thanks for the research. We'll do so then.
Nevertheless I'd consider it worthy to follow the proposed steps to see what happens.

Nightwalker-87, 2023-06-19:
@WRansohoff @jakeypoo @Mk-arc @nemtsov @mynameisdaniil
Please try to do the tests within the upcoming 2 weeks, if possible, in order to proceed.

Mk-arc, 2023-06-23:
Hi all. 

I did some testing. First of all my Setup:

[image]

CPU: STM32L063R8Tx with following connections:

Debugger:
[image]

uC:
[image]

First of all. I double checked that its possible to download my program using a J-Link debugger. Which works fine. Then furthermore, I tested with the ST-Link V2 in the picture above using STM32CubeProgrammer, which also works fine. Thus, I   am pretty confident that the Hardware is fine.

Now to st-link:
I checked out: d4b53b06b6ed8906798201ece463ffd2704051ef
```
 ./st-flash --version
v1.7.0-294-gd4b53b0
  ```
  ```
./st-info  --probe
/usr/local/share//stlink/chips: No such file or directory
Found 1 stlink programmers
  version:    V2J29S7
  serial:     680056000A0000385134534E
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x417
  ```

This is interesting now... It seems to not proberly detect the chip when I run the same with st-link installed from the ubuntu package repository I get different results:

```
st-info --version
v1.7.0
  ```
  ```
 st-info  --probe
Found 1 stlink programmers
  version:    V2J29S7
  serial:     680056000A0000385134534E
  flash:      65536 (pagesize: 128)
  sram:       8192
  chipid:     0x0417
  descr:      L0x3

  ```
  
  Not sure how to continue with the actual flashing test. Since its failing now even earlier?

Nightwalker-87, 2023-06-23:
@Mk-arc I don't see this on the F1, F4, G4 boards when using the same programmer (ST-Link v2 Clone) I have, so it can not be a general problem. However I noticed that your Programmer-Firmware is quite old. You may want to update it with the official STM32CubeProgrammer...

Mk-arc, 2023-06-23:
@Nightwalker-87 

Thanks for the quick response. I updated the FW:

Repo-version
```
st-info  --probe
Found 1 stlink programmers
  version:    V2J40S7
  serial:     680056000A0000385134534E
  flash:      65536 (pagesize: 128)
  sram:       8192
  chipid:     0x0417
  descr:      L0x3
 ```
 
 Fresh compiled:
 ```
./st-info --probe
/usr/local/share//stlink/chips: No such file or directory
Found 1 stlink programmers
  version:    V2J40S7
  serial:     680056000A0000385134534E
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x417
```

There is nothing relating to st-link in /usr/local/share. Also make uninstall does not work:

```
CMake Error at cmake/cmake_uninstall.cmake:2 (message):
  Cannot find install manifest:
  <name>/git/stlink/build/install_manifest.txt

make[3]: *** [CMakeFiles/uninstall.dir/build.make:70: CMakeFiles/uninstall] Error 1
make[2]: *** [CMakeFiles/Makefile2:394: CMakeFiles/uninstall.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:401: CMakeFiles/uninstall.dir/rule] Error 2
make: *** [Makefile:267: uninstall] Error 

```

Nightwalker-87, 2023-06-23:
@Mk-arc Something is not correct here. Make sure you don't have repo version and compiled variant installed at the same time - this may likely not work. I'd recommend to uninstall the package version, then delete the /build folder in your local project and restart compiling: `sudo make clean && sudo make install && sudo make uninstall`. Then ensure that there is no folder `stlink` in `/usr/local/share` and that there is nothing related left in `/usr/local/bin`, `/usr/local/include` and `/usr/local/lib`. Then run `sudo make clean && sudo make install`. --> What happens?

Mk-arc, 2023-06-23:
@Nightwalker-87 

I uninstalled the package version and did what you recommend.

Now its different but still not okay:

First of all. for some reason I can now not access the device without root.
```
./st-info --probe
2023-06-23T22:52:31 ERROR usb.c: Could not open USB device 0x0483:0x3748, access error.
Found 0 stlink programmers

```
And its still not properly detecting the probe.
```
sudo ./st-info --probe
Found 1 stlink programmers
  version:    V2J40S7
  serial:     680056000A0000385134534E
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x417
```

Nightwalker-87, 2023-06-23:
Packages `libusb-1.0` and `libusb-1.0-0-dev` are present?

Mk-arc, 2023-06-23:
Yes. I will try tomorrow late evening on another computer. Maybe something is skewed in my system.

Nightwalker-87, 2023-06-23:
Check /doc/compiling.md also --> Section: Set device access permissions and the role of udev

Mk-arc, 2023-06-24:
@Nightwalker-87 thanks for all the help. I now retried at another computer and there the probe gets detected properly and I can now reproduce the actual problem we are discussing in this issue:

```
~/git/stlink/build/bin/st-flash write ./STM32L0Breakout.bin 0x08000000
st-flash 1.7.0-11-ge662da0
2023-06-24T14:48:36 INFO common.c: L0x3: 8 KiB SRAM, 64 KiB flash in at least 128 byte pages.
file ./STM32L0Breakout.bin md5 checksum: e730ae43d067d1d3c98cffce13b5f8, stlink checksum: 0x0015e02c
2023-06-24T14:48:36 INFO common.c: Attempting to write 16056 (0x3eb8) bytes to stm32 address: 134217728 (0x8000000)
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000000 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000080 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000100 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000180 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000200 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000280 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000300 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000380 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000400 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000480 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000500 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000580 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000600 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000680 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000700 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000780 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000800 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000880 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000900 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000980 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000a00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000a80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000b00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000b80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000c00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000c80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000d00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000d80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000e00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000e80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000f00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08000f80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001000 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001080 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001100 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001180 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001200 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001280 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001300 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001380 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001400 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001480 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001500 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001580 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001600 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001680 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001700 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001780 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001800 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001880 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001900 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001980 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001a00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001a80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001b00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001b80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001c00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001c80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001d00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001d80 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001e00 erased
2023-06-24T14:48:36 INFO common.c: Flash page at addr: 0x08001e80 erased
[... 65 lines cut ...]
2023-06-24T14:48:36 INFO common.c: Starting Flash write for L0
2023-06-24T14:48:36 INFO common.c: Starting Half page flash write for STM32L core id
2023-06-24T14:48:36 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-06-24T14:48:36 INFO flash_loader.c: Clear DFSR
[... cut ...]
After the last entry the program freezes and has to be terminated.

Interestingly after this the ST-Link V2 clone probe was somehow in a broken state. Even after power cycle it was not being detected properly. Just after reuploading new FW with STM32CubeProgrammer it started working again.

``` 
~/git/stlink/build/bin/st-info --probe
Found 1 stlink programmers
  version:    V2J40S7
  serial:     680056000A0000385134534E
  flash:      65536 (pagesize: 128)
  sram:       8192
  chipid:     0x0417
  descr:      L0x3
``

Nightwalker-87, 2023-06-24:
Ok, good, so now try to apply the patch from here:

> **b)** @mattgbio on Dec 6, 2021: --> found that commenting-out the following lines in `src/common.c:3346` :
> 
> ```
> if (len > pagesize) {
>   if (stm32l1_write_half_pages(sl, addr, base, len, pagesize) == -1) {
>     // this may happen on a blank device!
>     WLOG("\nwrite_half_pages failed == -1\n");
>   } else {
>     off = (size_t)(len / pagesize) * pagesize;
>   }
> }
> ```

Nightwalker-87, 2023-06-24:
> st-flash 1.7.0-11-ge662da0
--> You have checked out the `master` branch instead of `develop`. 
Run `sudo make uninstall && sudo make clean`, then switch to develop, and retry...

Mk-arc, 2023-06-24:
tested with develop branch. WITHOUT applying the patch.

```
~/git/stlink/build/bin/st-flash write ./STM32L0Breakout.bin 0x08000000
st-flash 1.7.0-294-gd4b53b0
2023-06-24T21:14:05 INFO common.c: STM32L0xxx_Cat_3: 8 KiB SRAM, 64 KiB flash in at least 128 byte pages.
file ./STM32L0Breakout.bin md5 checksum: e730ae43d067d1d3c98cffce13b5f8, stlink checksum: 0x0015e02c
2023-06-24T21:14:05 INFO common_flash.c: Attempting to write 16056 (0x3eb8) bytes to stm32 address: 134217728 (0x8000000)
-> Flash page at 0x8003e80 erased (size: 0x80)
2023-06-24T21:14:06 INFO flash_loader.c: Starting Flash write for L0
2023-06-24T21:14:06 INFO flash_loader.c: Successfully loaded flash loader in sram
2023-06-24T21:14:06 INFO flash_loader.c: Clear DFSR
2023-06-24T21:14:06 ERROR flash_loader.c: write_buffer_to_sram() == -1
2023-06-24T21:14:06 WARN flash_loader.c: Failed to use flash loader, fallback to soft write
250/250 halfpages written
2023-06-24T21:14:07 INFO common_flash.c: Starting verification of write complete
2023-06-24T21:14:07 INFO common_flash.c: Flash written and verified! jolly good!
``

Nightwalker-87, 2023-06-24:
@Mk-arc Thanks, we are making progress. :ok_hand: .
Let me analyse the remaining inconsistency.

[4 more replies]

closed 2023-09-01 by Nightwalker-87