stlink-org/stlink #682: License of flashloaders source files

bluca, 2018-03-10:
Hi,

I would like to upload stlink to Debian. Before I can do that, I noticed that only 2 of the 8 files in flashloaders/ have a license - what about the others? Are they under the project-wide BSD-3-clause?

xor-gate, 2018-03-10:
Thats a very good question. There indeed a few bootloaders which are taken from the OpenOCD project under GPL. The others are based on the application note of the manifacturer, so they are part of the project which is indeed licensed as BSD-3-clause. If it is not clear you could propose a patch (PR) with correct headings. Thanks for the effort!

bluca, 2018-03-12:
So only flashloaders/stm32l0x.s and flashloaders/stm32lx.s are marked as GPL-2+ (I'll send a PR later to clarify the bit in the README.md), from the git log or the files I cannot say if any other are not BSD licensed - do you know if those 2 are the only exceptions?

xor-gate, 2020-04-09:
@Nightwalker-87 we need to update LICENSE and add the GPL part also for those files. I had some discussion in an email with the original authors which seemed to be copied from the OpenOCD project into stlink project. So we have a mix of GPL van BSD licensed code. If something is unclear please post a comment.

xor-gate, 2020-04-09:
Also @Nightwalker-87 it seems this issue is floating in the v1.5.1 milestone where 3 tickets are still open, I move them to the new milestone and next milestone so we sort out those things :=)

xor-gate, 2020-04-09:
I moved the issue to milestone v1.6.1 which is easy to fix. Also closed dangling v1.5.1 which was already release by me long time ago.

Nightwalker-87, 2020-04-09:
Yes, ok, i'll look at it. Thx.

Nightwalker-87, 2020-04-09:
> I moved the issue to milestone v1.6.1 which is easy to fix. Also closed dangling v1.5.1 which was already release by me long time ago.

No, please don't close any milestones which are open, this is on purpose as I am still reviewing old leftovers. Otherwise I can't keep on track with this.

Nightwalker-87, 2020-04-09:
We'll open a new ticket for this instead of recycling this old issue.

commit e9fe4e51 (2018-03-16): Debian packaging, CMake and README.md fixes (#683)

* Use https for debian/copyright

* Update dates and fix stanzas in debian/copyright

No need to have 2 stanzas for the same license

* Add BSD-3-clause body in debian/copyright

* Remove generated boilerplate from debian/rules

* Enable compiler hardening flags in debian/rules

* Fix source package name according to repo and docs

* Set source package section to electronics

See https://packages.debian.org/unstable/electronics/ for a full list

* Add long descriptions to packages

* Add gbp.conf file

* Update changelog for 1.5.0 release

*  [... cut]
closed 2020-04-09 by Nightwalker-87