stlink-org/stlink #685: STM32F303K8 detected as F334 device

WRansohoff, 2018-03-17:
- [X] Programmer/board type: Stlink/v2
- [X] Programmer firmware version: Unknown; generic ST-Link clone.
- [X] Operating system: Ubuntu 16.04
- [X] Stlink tools version and/or git commit hash: v1.5.0 release
- [X] Stlink commandline tool name: st-util
- [X] Target chip (and optional board): STM32F303K8

A as-detailed description possible of the problem with debug output when available.

Output:

```
st-util 1.5.0
2018-03-16T22:54:36 INFO usb.c: -- exit_dfu_mode
2018-03-16T22:54:36 INFO common.c: Loading device parameters....
2018-03-16T22:54:36 INFO common.c: Device connected is: F334 device, id 0x10016438
2018-03-16T22:54:36 INFO common.c: SRAM size: 0x3000 bytes (12 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
2018-03-16T22:54:36 INFO gdb-server.c: Chip ID is 00000438, Core ID is  2ba01477.
2018-03-16T22:54:36 INFO gdb-server.c: Listening at *:4242...
```

Expected/description:
The chips are STM32F303K8T6s, not F334s. I first noticed this with some chips that I ordered from Taobao and figured I'd just gotten some rejects from a mislabeled batch or something, but this also happens with chips from Digikey.

I don't think this causes any serious issues, but it's a little odd.

xor-gate, 2018-03-17:
The output is based on this:

https://github.com/texane/stlink/blob/master/include/stlink/chipid.h#L55

Probably the chipid 0x438 is F3xx series and the string should be different. As done by the pystlink project https://github.com/pavelrevak/pystlink/blob/master/lib/stm32devices.py#L360-L373

WRansohoff, 2018-03-18:
Interesting, thanks - I think I also have a Nucleo board with an F303K chip, I'll double-check that and/or look for reference documents to see if a new ID might make sense.

WRansohoff, 2019-01-13:
Sorry to leave this open for so long; I looked it up in RM0316, the ID is for F303x6, F303x8, F334, and F328 devices.

So I'm not sure what to call that or if it's even worth updating, but:

https://github.com/texane/stlink/pull/758

(Or just close the issue, I guess)

xor-gate, 2019-01-13:
PR #758 is merged, so probably you can close this issue now?

xor-gate, 2019-01-13:
@WRansohoff nice to know is when you use special text in commit messages it automaticly is added to the issue e.g (see issue #685). See https://help.github.com/articles/closing-issues-using-keywords/

WRansohoff, 2019-01-13:
Oh, good to know - thanks

closed 2020-03-18 by Nightwalker-87