stlink-org/stlink #689: Static executable for Linux

Zensey, 2018-03-21:
Is there any way to make a static executable ?

Vascom, 2018-03-21:
Why you need it?

Zensey, 2018-03-22:
ok.
may be I should consider delivery by means of packaging.

slyshykO, 2018-03-22:
> What do you mean?
> Может перейдём на русский?

I think that is possible make a static build for non-GUI part of stlink.
The only need is to add a static build for libusb and make necessary
changes in CMakeLists.txt. For now, it is impossible.

xor-gate, 2018-03-26:
Under linux there is no reason to have a static executable. You should get it from the distro repository or compile and install yourself. There is always a never ending struggle with static and non-static builds. Distro's don't like to build static (takes up more disk space). 

The static magic is only happening for OS X and Windows builds (for now). See

https://github.com/texane/stlink/blob/master/CMakeLists.txt#L132-L135
https://github.com/texane/stlink/blob/master/CMakeLists.txt#L169

Nightwalker-87, 2020-02-21:
Feedback closes #689.

closed 2020-02-21 by Nightwalker-87