stlink-org/stlink #690: Can't flash STM32L072

Faduf, 2018-03-21:
I have a similar problem than the issues #593 and #579. I can't flash my STM32L0 anymore and I have the failure "l1_stlink_flash_loader_run(0x8000000) failed! == -1".

- [X] Stlink/v2-1 onboard
- [X] Don't know, never changed, board bought in February 2018
- [X] Linux Ubuntu 16.04
- [X] st-flash version V1.5.0
- [X] st-flash
- [X] STM32L072CZ on the board "STM32L0 Discovery kit LoRa, Sigfox, low-power wireless"

```
st-flash write obj/l072.bin 0x8000000
st-flash 1.5.0
2018-03-21T19:35:00 INFO common.c: Loading device parameters....
2018-03-21T19:35:00 INFO common.c: Device connected is: L0x Category 5 device, id 0x20086447
2018-03-21T19:35:00 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2018-03-21T19:35:00 INFO common.c: Ignoring 96 bytes of 0x00 at end of file
2018-03-21T19:35:00 INFO common.c: Attempting to write 57100 (0xdf0c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0800df00 erased
2018-03-21T19:35:02 INFO common.c: Finished erasing 447 pages of 128 (0x80) bytes
2018-03-21T19:35:02 INFO common.c: Starting Half page flash write for STM32L core id
2018-03-21T19:35:02 INFO flash_loader.c: Successfully loaded flash loader in sram
2018-03-21T19:35:05 ERROR flash_loader.c: flash loader run error
2018-03-21T19:35:05 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2018-03-21T19:35:05 WARN common.c: 
write_half_pages failed == -1
445/446 pages written
2018-03-21T19:35:57 INFO common.c: Starting verification of write complete
2018-03-21T19:35:57 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```

xor-gate, 2018-03-26:
Thats a pitty, the flash loaders for the L0 and L1 seem to be unstable. In the meantime you should try to use openocd if that is possible to get a more stable result.

Faduf, 2018-03-26:
Ok, thanks for the feedback. Openocd works fine. Don't hesitate if you need help to debug.

Faduf, 2018-04-01:
This is working in boot mode (boot to Vdd). I'm closing the issue, thanks.

xor-gate, 2018-04-01:
Great it works now

Nightwalker-87, 2020-03-14:
@Faduf: Does this work without openocd as well with Release v1.6.0 ? Can someone verify?

Faduf, 2020-03-21:
@Nightwalker-87 sorry for not giving feedback on this topic but I'm stuck at home due to the coronavirus pandemy. My boards with the F0 chip are at the office and I can't go there until few weeks.

Nightwalker-87, 2020-03-21:
@Faduf: No problem at all, we're not in a hurry. Thx for your feedback. :+1:

Nightwalker-87, 2020-04-02:
Sorry for the delay. Please check if #876 resolves this problem as it resolves #579, one of the similar issues you mentioned in your opening post.

grevaillot, 2020-04-03:
I just checked on my tree with the same board (l0 discovery lora thing), apparently there's a flash loader run error - but the flashing goes well.. i dont think i have an other l0 device to check.

>  % ./build/Release/st-flash  write /tmp/saved.bin 0x8000000        
> st-flash 1.6.0-98-gd7cffeb-dirty
> 2020-04-03T10:55:23 INFO common.c: L0xx Category 5: 20 KiB SRAM, 192 KiB flash in 128 byte pages.
> 2020-04-03T10:55:23 INFO common.c: Attempting to write 196608 (0x30000) bytes to stm32 address: 134217728 (0x8000000)
> Flash page at addr: 0x0802ff80 erased
> 2020-04-03T10:55:33 INFO common.c: Finished erasing 1536 pages of 128 (0x80) bytes
> 2020-04-03T10:55:33 INFO common.c: Starting Half page flash write for STM32L core id
> 2020-04-03T10:55:33 INFO flash_loader.c: Successfully loaded flash loader in sram
> 2020-04-03T10:55:34 ERROR flash_loader.c: flash loader run error
> 2020-04-03T10:55:34 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
> 2020-04-03T10:55:34 WARN common.c: 
> write_half_pages failed == -1
> 1535/1536 pages written
> 2020-04-03T10:58:38 INFO common.c: Starting verification of write complete
> 2020-04-03T10:58:40 INFO common.c: Flash written and verified! jolly good!

I'll have a look at the flash loader run error.

Nightwalker-87, 2020-04-03:
Ah, thx @grevaillot! I know about this error from quite a few other tickets.
It would be very helpful indeed, if you could find a solution to this.

Nightwalker-87, 2020-05-26:
@grevaillot: Any update on this?

PR #1113 (2021-03-25): Flash loader rework
In this PR, the waiting for data writing is changed from cycles to ms. This should be a more robust solution. 
Fixed a flash loader bug. The flash loader had an incorrect busy wait cycle. The use of bootloader argument registers during boot removed. Removed magic with recalculation of the `size` to the `count`. The number of bytes to load (`size`) is transferred to the flash loader now.
Added disabling of interrupts during flash loader execution. In some cases interrupts caused a failure

(Closes #356, Closes #556, Closes #593, Closes #597, Closes #607, Closes #612, Closes  [... cut]
closed 2021-03-25 by Nightwalker-87