stlink-org/stlink #693: SHA-256 mismatch for 1.5.0.tar.gz

ilovezfs, 2018-03-26:
Homebrew users currently cannot build stlink from source due to a SHA-256 checksum mismatch for 1.5.0.tar.gz (https://github.com/texane/stlink/archive/1.5.0.tar.gz)

Fri Feb 16 06:26:57 2018 -0800 the SHA-256 was
ce26b3db73f0bdf87ced78a2a90f3d515914f7d1211b1ca4b9acdf1882ca9d81

Now it's changed to
4f0301edf9e904688ba31cf181653102f367c747e23ad356de1c430d1a654e2e

Do you have any idea what happened here? To rule out malicious circumstances, we'll need to know why the file has changed before the checksum can be updated in Homebrew. Thanks!

Vascom, 2018-03-26:
$ diff -uNdr rpmbuild/SOURCES/stlink-1.5.0 stlink-1.5.0
```
diff -uNdr rpmbuild/SOURCES/stlink-1.5.0/doc/tested-boards.md stlink-1.5.0/doc/tested-boards.md
--- rpmbuild/SOURCES/stlink-1.5.0/doc/tested-boards.md  2018-02-16 15:06:36.000000000 +0300
+++ stlink-1.5.0/doc/tested-boards.md   2018-02-19 22:00:29.000000000 +0300
@@ -47,6 +47,7 @@
 * Nucleo-L152RE board
 * [Nucleo-L476RG board](http://www.st.com/en/evaluation-tools/nucleo-l476rg.html)
 * STM32L452RET6 (STM32 Nucleo-L452RE board)
+* STM32F042K6 (STM32 Nucleo-32 Board)
 
 Please report any and all known working combinations so I can update this!
 
diff -uNdr rpmbuild/SOURCES/stlink-1.5.0/README.md stlink-1.5.0/README.md
--- rpmbuild/SOURCES/stlink-1.5.0/README.md     2018-02-16 15:06:36.000000000 +0300
+++ stlink-1.5.0/README.md      2018-02-19 22:00:29.000000000 +0300
@@ -3,7 +3,7 @@
 
 [![GitHub release](https://img.shields.io/github/release/texane/stlink.svg)](https://github.com/texane/stlink/releases/latest)
 [![BSD licensed](https://img.shields.io/badge/license-BSD-blue.svg)](https://raw.githubusercontent.com/hyperium/hyper/master/LICENSE)
-[![GitHub commits](https://img.shields.io/github/commits-since/texane/stlink/1.4.0.svg)](https://github.com/texane/stlink/compare/1.4.0...master)
+[![GitHub commits](https://img.shields.io/github/commits-since/texane/stlink/1.5.0.svg)](https://github.com/texane/stlink/compare/1.4.0...master)
 [![Downloads](https://img.shields.io/github/downloads/texane/stlink/total.svg)](https://github.com/texane/stlink/releases)
 [![Linux Status](https://img.shields.io/travis/texane/stlink/master.svg?label=linux)](https://travis-ci.org/texane/stlink)
 [![Build Status](https://jenkins.ncrmnt.org/buildStatus/icon?job=GithubCI/stlink)](https://jenkins.ncrmnt.org/job/GithubCI/job/stlink/)
diff -uNdr rpmbuild/SOURCES/stlink-1.5.0/.version stlink-1.5.0/.version
--- rpmbuild/SOURCES/stlink-1.5.0/.version      2018-02-16 15:06:36.000000000 +0300
+++ stlink-1.5.0/.version       2018-02-19 22:00:29.000000000 +0300
@@ -1 +1 @@
-1.4.0
+1.5.0
```

xor-gate, 2018-03-26:
Good catch, I had some problems with tagging/releasing and needed to fixup some things. Current shasum is the one which is correct. There has not been tampered with in any way (as myself and the other members).

closed 2018-03-26 by xor-gate

MNS003, 2018-05-21:
I got the same error, how can I fix it? I am running homebrew on linux (ubuntu).