stlink-org/stlink #698: Unable to build on ubuntu

protodmorgan, 2018-03-28:
I'm attempting to install this on ubuntu. However, it seem I am missing something.  when I follow the instructions make release. I am hitting the following error:

[RELEASE]
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:27: recipe for target 'release' failed
make: *** [release] Error 2

xor-gate, 2018-03-28:
You should run `make clean` because CMake didn't generated any makefiles. Make sure you have installed `cmake libusb-dev` or else it wont build.

protodmorgan, 2018-03-28:
results in the same issue. 

 sudo apt-get install cmake libusb-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libusb-dev is already the newest version (2:0.1.12-28).
cmake is already the newest version (3.5.1-1ubuntu3).
0 upgraded, 0 newly installed, 0 to remove and 108 not upgraded.

@DESKTOP-SLOJ0HQ:~/stlink$ make
[RELEASE]
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:27: recipe for target 'release' failed
make: *** [release] Error 2

slyshykO, 2018-03-28:
You should install libusb-1.0-dev( not sure that is correct name) but pack
may contain libusb-1.0

protodmorgan, 2018-03-28:
> You should install libusb-1.0-dev( not sure that is correct name) but pack
>may contain libusb-1.0

even with that installed Same issue

slyshykO, 2018-03-28:
Ok, I realize that makefile or instruction is outdated. Or both. You can
build with Cmake, or wait for someone fix makefile.

xor-gate, 2018-03-28:
You need to make sure the `build` folder is removed when the cmake run failed which `make clean` does.

slyshykO, 2018-03-29:
I successfully build the stlink following instruction.
And also in Orange Pi Zero, that relies on #621 .

protodmorgan, 2018-03-29:
thanks..as I said I was probably missing something. trying to learn.

protodmorgan, 2018-03-29:
Seem to run into a problem now after compiling, I am running st-info etc, and I get. st-info: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory 

can confirm those libs exist but are in lib/lib, and I did run ldconfig

Vascom, 2018-03-29:
It is bug #700

fedorg, 2018-05-18:
run ldconfig

jjmilburn, 2019-02-07:
Running `ldconfig` did not resolve this for me. I had to first manually copy the files `libstlink.so.1` / `libstlink.so` to `/usr/bin/lib`.

Nightwalker-87, 2020-02-21:
Please verify if the problem still exists in Release v1.6.0.

Basilisvirus, 2020-03-13:
Happened to me when i downloaded Source code (tar.gz).

Then, i downloaded  Source code (zip) and it worked.

Nightwalker-87, 2020-03-13:
@Basilisvirus: Interesting. Nevertheless both archive files .zip and .tar.gz. are generated automatically by github. Are there any local problems with unzipping .tar.gz or directory paths?

Basilisvirus, 2020-03-13:
@Nightwalker-87 Thats what i am thinking. Maybe my download was not completed/corrupted or there was an unzip error.

Nightwalker-87, 2020-03-13:
Could you retry the .tar.gz archive to verify, so that we can finally resolve this issue with the knowledge that both work correctly? That would be great.

Basilisvirus, 2020-03-13:
I would, but right now i get issue #478 error so let me fix that first xD

Nightwalker-87, 2020-03-13:
ok, of course we welcome any helpful contribution. :+1:

Basilisvirus, 2020-03-13:
I re-installed the release 1.6.0, using the **Source code (tar.gz)** and it installed correctly, so there is no issue here.
Plus, i installed it using `sudo make` instead of `make` and all my other installation-based problems were solved as well.

Nightwalker-87, 2020-03-13:
Yes we can! :+1: :1st_place_medal:

Nightwalker-87, 2020-03-20:
@protodmorgan: Which version did you originally use? Does it already work with v1.5.1? This would be good to know to allocate this ticket to the correct version in the changelog. I am trying to find the related fix for it.

Nightwalker-87, 2020-04-12:
> Running `ldconfig` did not resolve this for me. I had to first manually copy the files `libstlink.so.1` / `libstlink.so` to `/usr/local/lib/`. Only then did `ldconfig` make any difference.

@slyshykO: Was this fixed in PR 701? Could you have a look at this?

slyshykO, 2020-04-12:
@Nightwalker-87  #701 not about this issue.

Nightwalker-87, 2020-04-12:
Then I don't know either. Does anyone else have an idea on this?
The rest of the issue seems to have resolved from fixing local issues.

Nightwalker-87, 2020-06-03:
I'd need someone with an ubuntu system for assistance with testing for a final verification with older releases. If this was a general issue on Ubuntu, then I believe this has been fixed in the meanwhile - otherwise it would have derived from a local circumstance we can not reproduce.

Nightwalker-87, 2020-06-13:
Final review of this issue:
1) `make[1]: *** No targets specified and no makefile found. Stop.`
--> Solution: https://github.com/stlink-org/stlink/issues/698#issuecomment-377037857
2) `st-info: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory`
--> Solution: https://github.com/stlink-org/stlink/issues/698#issuecomment-598891035

As there are no remaining open points to address, this issue is now finally closed.

closed 2020-06-13 by Nightwalker-87