stlink-org/stlink #700: Installation issue under linu

cedricboudinet, 2018-03-29:
Hello,
stlink installation under Linux is broken since modifications for #695: libstlink.so (and others) installs in /usr/local/lib/lib instead of /usr/local/lib
It seems that the error comes from these instructions:
 if(IS_DIRECTORY CMAKE_LIBRARY_PATH)
 else()
    set(CMAKE_LIBRARY_PATH "lib")
 endif()
If I change it to 
 if(IS_DIRECTORY CMAKE_LIBRARY_PATH)
 else()
    set(CMAKE_LIBRARY_PATH "")
 endif()
then it works again.
I can create a PR if you want, but I cannot test the modification under other OS.

Vascom, 2018-03-29:
I can confirm.
Also hardcode path to "lib" breake x86_64 with /usr/lib64 folder.

xor-gate, 2018-03-29:
Hi guys, I agree with your both. We must find a working solution for all cases (with less cmake code).

slyshykO, 2018-03-29:
In my defence, I would say that I encounter a situation when CMAKE_LIBRARY_PATH is a list of paths. That broke all install command completely.
In my opinion, we should remove CMAKE_LIBRARY_PATH  from all install commands.

Cheng-SG, 2018-03-30:
I encountered the same issue

Vascom, 2018-03-30:
@Cheng-SG please try #701 fix

commit 88935336 (2018-04-16): fix #700 (#701)

* first try to fix #700
Use CMAKE_INSTALL_PREFIX for install paths

* check IS_DIRECTORY LIB_INSTALL_DIR
* fix incorrect VAR usage in condition
* changed debian/rules according to @bluca patch
closed 2020-03-18 by Nightwalker-87