stlink-org/stlink #703: Issue with libusb 1.0.22 / function 'libusb_set_debug' is deprecated

LeisureLadi, 2018-04-01:
Compiling stlink with 'make release' and 'make debug' with libusb 1.0.22 currently throws these warning messages:

...
Scanning dependencies of target stlink
[  3%] Building C object CMakeFiles/stlink.dir/src/chipid.c.o
[  6%] Building C object CMakeFiles/stlink.dir/src/common.c.o
[ 10%] Building C object CMakeFiles/stlink.dir/src/usb.c.o
[ 13%] Building C object CMakeFiles/stlink.dir/src/sg.c.o
/home/pi/stlink/src/sg.c: In function ‘stlink_open’:
/home/pi/stlink/src/sg.c:957:5: warning: ‘libusb_set_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]
     libusb_set_debug(slsg->libusb_ctx, 3);
     ^~~~~~~~~~~~~~~~
In file included from /home/pi/stlink/include/stlink/sg.h:17:0,
                 from /home/pi/stlink/include/stlink.h:229,
                 from /home/pi/stlink/src/sg.c:85:
/usr/include/libusb-1.0/libusb.h:1300:18: note: declared here
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~
...
Scanning dependencies of target stlink-static
[ 43%] Building C object CMakeFiles/stlink-static.dir/src/chipid.c.o
[ 46%] Building C object CMakeFiles/stlink-static.dir/src/common.c.o
[ 50%] Building C object CMakeFiles/stlink-static.dir/src/usb.c.o
[ 53%] Building C object CMakeFiles/stlink-static.dir/src/sg.c.o
/home/pi/stlink/src/sg.c: In function ‘stlink_open’:
/home/pi/stlink/src/sg.c:957:5: warning: ‘libusb_cc1: all warnings being treated as errors
CMakeFiles/stlink.dir/build.make:134: die Regel für Ziel „CMakeFiles/stlink.dir/src/sg.c.o“ scheiterte
make[3]: *** [CMakeFiles/stlink.dir/src/sg.c.o] Fehler 1
CMakeFiles/Makefile2:141: die Regel für Ziel „CMakeFiles/stlink.dir/all“ scheiterte
make[2]: *** [CMakeFiles/stlink.dir/all] Fehler 2
Makefile:149: die Regel für Ziel „all“ scheiterte
make[1]: *** [all] Fehler 2
Makefile:27: die Regel für Ziel „release“ scheiterte
make: *** [release] Fehler 2
_debug’ is deprecated: Use libusb_set_option instead [-Wdeprecated-declarations]
     libusb_set_debug(slsg->libusb_ctx, 3);
     ^~~~~~~~~~~~~~~~
In file included from /home/pi/stlink/include/stlink/sg.h:17:0,
                 from /home/pi/stlink/include/stlink.h:229,
                 from /home/pi/stlink/src/sg.c:85:
/usr/include/libusb-1.0/libusb.h:1300:18: note: declared here
 void LIBUSB_CALL libusb_set_debug(libusb_context *ctx, int level);
                  ^~~~~~~~~~~~~~~~

Should I revert back to earlier releases of libusb and if so which one?

Thank you,

slyshykO, 2018-04-01:
Hi @LeisureLadi  .
I've added PR with a fix. Please check it.

LeisureLadi, 2018-04-01:
Problem solved! Compiles now without errors. Great.

However, 'libstlink.so.1' was missing initially. I'm not sure, if the libraries are correctly saved in '/usr/local/lib/lib'. I changed the directory name to '/usr/local/lib/stlink' and added this path to '/etc/ld.so.conf.d/libc.conf'. After running 'sudo ldconfig', all is fine now.

Thanks a lot for the quick reply!
Happy Easter!

xor-gate, 2018-04-01:
Thanks guys for catching up

commit aaf8e920 (2018-04-01): Fix libusb function deprecation used for stlinkv1

Fixes #703
closed 2018-04-01 by xor-gate