stlink-org/stlink #705: Can't flash STM32F030F4P6

Keziolio, 2018-04-05:
- [X] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: Don't know
- [X] Operating system: Arch Linux
- [X] Stlink tools version: just compiled from git
- [X] Stlink commandline tool name:  `st-flash`
- [X] Target chip: STM32F030F4P6

```
 ./st-flash  --reset write firmware.bin     0x8000000
st-flash 1.4.0-32-gaaf8e92
2018-04-05T23:36:50 INFO common.c: Loading device parameters....
2018-04-05T23:36:50 INFO common.c: Device connected is: F0 small device, id 0x10006444
2018-04-05T23:36:50 INFO common.c: SRAM size: 0x1000 bytes (4 KiB), Flash: 0x4000 bytes (16 KiB) in pages of 1024 bytes
2018-04-05T23:36:50 INFO common.c: Attempting to write 5932 (0x172c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08001400 erased
2018-04-05T23:36:50 INFO common.c: Finished erasing 6 pages of 1024 (0x400) bytes
2018-04-05T23:36:50 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2018-04-05T23:36:50 INFO flash_loader.c: Successfully loaded flash loader in sram
2018-04-05T23:36:53 ERROR flash_loader.c: flash loader run error
2018-04-05T23:36:53 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1

```
```
./st-info --probe
Found 1 stlink programmers
 serial: 303030303030303030303031
openocd: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
  flash: 16384 (pagesize: 1024)
   sram: 4096
 chipid: 0x0444
  descr: F0 small device
```

I can't manage to flash one of these: https://www.ebay.com/itm/Mini-System-Development-Board-ARM-STM32-F030F4P6-CORTEX-M0-Core-32bit-48-MHz/331978736223?hash=item4d4b79825f:g:NncAAOSw-09aQKDm

It seems to --probe correctly, erase correctly, the stlink led flashes, but it does not flash the actual firmware, i tried to flash the same firmware with the same setup on an stm32f103c8t6 and it worked.
I'm using a cheap chinese stlink clone.

Output with --debug:
```
2018-04-05T23:14:00 DEBUG common.c: stlink current mode: mass
2018-04-05T23:14:00 DEBUG common.c: stlink current mode: mass
2018-04-05T23:14:00 DEBUG common.c: *** stlink_enter_swd_mode ***
2018-04-05T23:14:00 DEBUG common.c: *** looking up stlink version
2018-04-05T23:14:00 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2018-04-05T23:14:00 DEBUG common.c: stlink pid     = 0x3748
2018-04-05T23:14:00 DEBUG common.c: stlink version = 0x2
2018-04-05T23:14:00 DEBUG common.c: jtag version   = 0x11
2018-04-05T23:14:00 DEBUG common.c: swim version   = 0x4
2018-04-05T23:14:00 DEBUG common.c: *** stlink_jtag_reset ***
2018-04-05T23:14:00 DEBUG common.c: *** stlink_reset ***
2018-04-05T23:14:01 INFO common.c: Loading device parameters....
2018-04-05T23:14:01 DEBUG common.c: *** stlink_core_id ***
2018-04-05T23:14:01 DEBUG common.c: core_id = 0x0bb11477
2018-04-05T23:14:01 DEBUG common.c: *** stlink_read_debug32 0 is 0xe0042000
2018-04-05T23:14:01 DEBUG common.c: *** stlink_read_debug32 10006444 is 0x40015800
2018-04-05T23:14:01 DEBUG common.c: *** stlink_read_debug32 ffff0010 is 0x1ffff7cc
2018-04-05T23:14:01 INFO common.c: Device connected is: F0 small device, id 0x10006444
2018-04-05T23:14:01 INFO common.c: SRAM size: 0x1000 bytes (4 KiB), Flash: 0x4000 bytes (16 KiB) in pages of 1024 bytes
2018-04-05T23:14:01 DEBUG common.c: *** set_swdclk ***
2018-04-05T23:14:01 DEBUG common.c: stlink current mode: debug (jtag or swd)
2018-04-05T23:14:01 DEBUG common.c: stlink current mode: debug (jtag or swd)
2018-04-05T23:14:01 DEBUG common.c: *** stlink_force_debug_mode ***
2018-04-05T23:14:01 DEBUG common.c: *** stlink_status ***
2018-04-05T23:14:01 DEBUG common.c:   core status: halted
2018-04-05T23:14:01 INFO common.c: Attempting to write 5932 (0x172c) bytes to stm32 address: 134217728 (0x8000000)
2018-04-05T23:14:01 DEBUG common.c: *** stlink_core_id ***
2018-04-05T23:14:01 DEBUG common.c: core_id = 0x0bb11477
2018-04-05T23:14:01 DEBUG common.c: *** stlink_read_debug32 10 is 0x4002200c
[... 141 lines cut ...]
2018-04-05T23:14:04 DEBUG common.c: *** stlink_close ***

data_len = 2 0x2
 80 00

data_len = 2 0x2
 80 00

data_len = 2 0x2
 80 00

data_len = 2 0x2
 80 00

data_len = 2 0x2
 80 00

data_len = 2 0x2
 81 00

data_len = 2 0x2
 81 00

stlink_fwrite_flash() == -1
```

Keziolio, 2018-04-05:
Ok the issue was that the chip had the usual write protection enabled, I had to use a serial adapter and execute this other tool

```
stm32flash -k /dev/ttyUSB0
```

To unlock it.

So, maybe, this issue can be changed to "make st-flash give an appropriate warning on what's happening" because this was rather counter intuitive

xor-gate, 2018-04-06:
There are multiple issues closed and related to write protected microcontrollers which can be unprotected by mass erase and then flash. Not sure how we can detect this but it is a serious usability problem of the tools.

hggq, 2018-06-01:
my stm32f103   
stm32f1xx_hal_msp.c file
/**DISABLE: JTAG-DP Disabled and SW-DP Disabled 
    */
  __HAL_AFIO_REMAP_SWJ_ENABLE();

ebable sw is ok

now stm32CubeMX Create project    this  value   default Disabled

st-flash  write bin file not write in
2018-06-01T17:51:44 INFO common.c: Loading device parameters....
2018-06-01T17:51:44 WARN common.c: Invalid flash type, please check device declaration
Unknown memory region

joe-skb7, 2019-02-07:
@nit1995 In my case, next command helps:

    openocd -f interface/stlink-v2.cfg \
            -f target/stm32f1x.cfg \
            -c "init; reset halt; stm32f1x unlock 0; reset halt; exit"

Of course, you need to replace interface and target files with ones appropriate for your case.

joe-skb7, 2019-02-07:
@nit1995 Se my [post above](https://github.com/texane/stlink/issues/705#issuecomment-395981733).

korr237i, 2019-05-18:
> Just my two cents: had the same situation, only for STM32F1. Managed to disable write protection using STLinkV2 (which is integrated in STM32F1DISCOVERY board) via OpenOCD command:
> 
> ```
> $ openocd -f interface/stlink-v2.cfg \
>           -f target/stm32f1x.cfg \
>           -c "init; reset halt; stm32f1x unlock 0; reset halt; exit"
> ```
> 
> where `target` is your CPU or board.

Nightwalker-87, 2020-02-21:
Any update on this?

kahanchudasama, 2020-06-21:
> Just my two cents: had the same situation, only for STM32F1. Managed to disable write protection using STLinkV2 (which is integrated in STM32F1DISCOVERY board) via OpenOCD command:
>
> ```
> $ openocd -f [...]

HI, 
can you please guide me how can we achive level 2 protection through openocd

joe-skb7, 2020-06-21:
@kahanchudasama Locking with OpenOCD can be done as shown [here](http://openocd.org/doc/html/Flash-Commands.html) (in `Flash Driver: stm32f1x` section, see `lock` command). Something like this should work (if you use STLinkv2 programmer and your chip is STM32F1xx):

    $ openocd -f interface/stlink-v2.cfg \
              -f target/stm32f1x.cfg \
              -c "init; reset halt; stm32f1x lock 0; reset halt; exit"

Whether it's level 2 or no -- can't say for sure, but documentation for `lock` command says:

    Locks the entire stm32 device against reading.

I think it's better to read STM32 corresponding registers and check how exactly it was locked, after performing mentioned OpenOCD `lock` command. Either that, or check how that OpenOCD commands is implemented.

Nightwalker-87, 2020-06-21:
As @xor-gate mentioned this is a serious usability problem.
Can somebody propose a PR to finally fix this in the toolset?

Nightwalker-87, 2021-04-04:
IMHO documentation should be improved here.
Can one think of a reasonable way to remove write protection via the toolset directly?

Ant-ON, 2021-04-05:
@Nightwalker-87 Yes it possible through writing of protection bits (part of option bytes). 
I added check `WRPERR` error to the https://github.com/Ant-ON/stlink/tree/fl_fix branch. In the event of a write error due to the set protection bits, a corresponding message should be displayed.

PR #1124 (2021-04-13): Improvements and fixes of the flash loaders, unification of the reset function
STM32L4 flash loader: fixed waiting for busy
STM32F1 flash loader: fixed setting of the PG flag
STM32Lx flash loader: fixed size of the flash loader
Added disabling of the DMA channels and watchdog timers while flash loader are running.
All possible reset options are combined into one function. When using the JTAG reset, the use of the pulse command has been removed. Instead, separate pull-down and pull-up commands are used.
Added target change message to the gdb server.

(Closes #244, Closes #382, Closes #705, Clos [... cut]
closed 2021-04-13 by Nightwalker-87