stlink-org/stlink #713: GUI failed to parse .hex file

FoxyHawk, 2018-06-10:
- [x] Programmer/board type: Stlink/v2-onboard
- [x] Programmer firmware version: V2.J29.M18
- [x] Operating system: Arch Linux (Manjaro)
- [x] Stlink tools version and/or git commit hash: v1.5.0-2 (lastest Manjaro repo)
- [x] Stlink commandline tool name: `stlink-gui`
- [x] Target chip (and optional board): STM32F767ZI

Reading of .hex files in stlink-gui differ from Windows Official ST-link Utility.
It seems to failed to interpret the first line of the file as an address offset.
```
:020000040800F2
:1000000000000820290300082F030008330300081C
:100010007D03000881030008C903000800000000F8
:1000200000000000000000000000000011040008B3
:10003000150400080000000019040008150D000850
:100040002D0400082D0400082D0400082D040008CC
......
```
Based on https://en.wikipedia.org/wiki/Intel_HEX, the first line must be read as start at 0x08000000.
On Arch Linux, the file become unusable (but the programming  and writing are fine btw) and the result on the card is unpredictable.
[image]
[image]

On Windows the file is correctly parsed and the result is what I expect.
[image]
[image]

The .hex file in .txt (only a basic blink LED for my card):
[test2.txt](https://github.com/texane/stlink/files/2087654/test2.txt)

xor-gate, 2018-06-22:
@dev26th I would like you to have a look at this issue because you have integrated intel hex reading some time ago in https://github.com/texane/stlink/pull/459.

dev26th, 2018-06-24:
Sure, it's just a small piece of code

xor-gate, 2018-06-27:
Thanks for reporting and resolving!

closed 2018-06-27 by xor-gate