stlink-org/stlink #715: Support for STM32F103 RCT6

walterinho, 2018-06-18:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ ] Programmer/board type: Stlink/v2
- [ ] Programmer firmware version: V2J29S7 (I'm not sure if this is correct, I got it from the Windows ST-Link utility)
- [ ] Operating system: Linux Ubuntu 16.04
- [ ] Stlink tools version and/or git commit hash: v1.5.0
- [ ] Stlink commandline tool name: st-util
- [ ] Target chip (and optional board): STM32F103 RCT6

A as-detailed description possible of the problem with debug output when available.

Output:
I'm trying to flash an STM32F103 RCT6 from a Hoverboard's mainboard, but when I use the command st-util I get the following:

st-util 1.4.0-39-g6db0fc2
2018-06-18T13:04:46 INFO common.c: Loading device parameters....
2018-06-18T13:04:46 WARN common.c: unknown chip id! 0x5fa0004

I also tried with the MCU STM32F103 C8T6 and it works fine (I was able to flash the MCU and debug a simple code).

I'm a newbie, please bear with me if I missing some important data. Thanks.

Thank you,

xor-gate, 2018-06-22:
Weird, the STM32F103RCT6 is a MCU which is widely used and should work. Not sure why your C8T6 variant board works. Are you sure the hardware is correct? SWD correctly populated? Could you also paste the verbose output of the target which works? Normaly the chipid should be one of the following for F1 targets https://github.com/texane/stlink/blob/master/include/stlink/chipid.h#L16-L18 (which is according to the datasheets). It is used to detect which MCU is connected to the programmer (software) during probe.

walterinho, 2018-06-23:
Hi again,

I just got a new mainboard from another manufacturer and I was successfully able to load my code into the STM32F103RCT6. So, we can say that this is not a problem with this project.

But maybe you could please help me figure out what the problem could be or what should I do with the other board to load the code (it works perfectly on the hoverboard), before closing this thread.

[image]

Thanks again.

nicolai86, 2019-03-21:
I'm seeing the same issue but connecting to a proton-c, which uses the STM32F303CCT6.

xor-gate, 2019-03-23:
I have seen programming issues can be due to bad cabling (USB, SWD) and programmers which are broken.

electrictwat, 2019-04-21:
Hi, 
I'm using the cheap stlink programmer available on ebay/aliexpress. No issue with KEIL or Stlinkutility on WIndows. I got an error output with only the stlink plugged (the same output with the target board connected)

` INFO common.c: Loading device parameters....
 WARN common.c: unknown chip id! 0x5fa0004
`
One of the programmer has a STM32F103C8T6 and the other STM32F101C8T6. 
I'm using the  v1.5.0 .

COuld I modify the files linked here : https://github.com/texane/stlink/issues/673#issuecomment-367432128 and add references for the chip and recompile [?]

davidko-ww, 2019-06-17:
Hello, I have a STM32F3 Discovery kit for STM32F303 MCU which also has this problem. 

```
st-util 1.5.1-30-g84f63d2
2019-06-17T16:54:40 DEBUG common.c: stlink current mode: debug (jtag or swd)
2019-06-17T16:54:40 DEBUG common.c: stlink current mode: debug (jtag or swd)
2019-06-17T16:54:40 DEBUG common.c: *** looking up stlink version
2019-06-17T16:54:40 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2019-06-17T16:54:40 DEBUG common.c: stlink pid     = 0x374b
2019-06-17T16:54:40 DEBUG common.c: stlink version = 0x2
2019-06-17T16:54:40 DEBUG common.c: jtag version   = 0x1b
2019-06-17T16:54:40 DEBUG common.c: swim version   = 0xf
2019-06-17T16:54:40 DEBUG common.c: *** set_swdclk ***
2019-06-17T16:54:40 DEBUG common.c: *** stlink_jtag_reset ***
2019-06-17T16:54:40 DEBUG common.c: *** stlink_reset ***
2019-06-17T16:54:40 DEBUG common.c: *** stlink_write_debug32 5fa0004 to 0xe000ed0c
2019-06-17T16:54:40 INFO common.c: Loading device parameters....
2019-06-17T16:54:40 DEBUG common.c: *** stlink_core_id ***
2019-06-17T16:54:40 DEBUG common.c: core_id = 0x00000000
2019-06-17T16:54:40 DEBUG common.c: *** stlink_read_debug32 5fa0004 is 0xe0042000
2019-06-17T16:54:40 WARN common.c: unknown chip id! 0x5fa0004
2019-06-17T16:54:40 DEBUG common.c: *** stlink_close ***

```

Any ideas? This is a brand new kit, I'm fairly certain there is nothing wrong due to bad cabling...

xor-gate, 2019-06-19:
I'm not sure but the chipid is indeed incorrect, the coreid seems to read also 0x00000000. https://github.com/texane/stlink/blob/master/include/stlink/chipid.h#L13.

doobedoobedo, 2019-08-22:
I just had the same thing with a blue pill - my cheap stlink clone appears to have the markings for SWDIO and SWCLK swapped. After swapping the lines over it flashed fine.

sderr, 2019-09-27:
Thanks a lot doobedoobedo ! I just bumped into the same issue, lines are swapped.

Nightwalker-87, 2020-02-21:
Feedback resolves #715.

closed 2020-02-21 by Nightwalker-87