stlink-org/stlink #717: No makefile found

PirikitoOreo, 2018-06-23:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2018-06-23:
Please provide details.

PirikitoOreo, 2018-06-23:
Hello, after typing next command: make release
I've got next error:
[RELEASE]
make[1]: *** No targets specified and no makefile found. Stop.
Makefile:27: recipe for target 'release' failed
make: *** [release] Error 2

xor-gate, 2018-06-23:
Provide more details please. As the first post in this thread suggests.
You should remove the build folder probably due to non-installed depencies the Makefile is not generated under build/Releas/Makefile. I advice you to read https://github.com/texane/stlink/blob/master/doc/compiling.md.

PirikitoOreo, 2018-06-23:
Thank you for your reply, but iI think I'm completely lost.
Could you show me commands that i have to un please.

xor-gate, 2018-06-23:
Start reading https://github.com/texane/stlink/blob/master/doc/compiling.md.

xor-gate, 2018-06-23:
You are not answering my questions and providing details, so I can not help you.

xor-gate, 2018-06-23:
I think you forgot to install cmake.

PirikitoOreo, 2018-06-23:
After installing **build-essential**, commands **make release** and **make debug** worked well.
What sould I do next?
Thanks

xor-gate, 2018-06-24:
You should build a debian package with `make package`, and install the package with `dpkg -i *.deb`.

PirikitoOreo, 2018-06-24:
Thank you for your reply. I executed your commands an I think everything is OK.
Bellow I post the output:
**~/stlink.git $ make package
[PACKAGE] Release
[ 23%] Built target stlink-static
[ 46%] Built target stlink
[ 56%] Built target st-flash
[ 63%] Built target st-info
[ 76%] Built target st-util
[ 83%] Built target sg
[ 90%] Built target usb
[100%] Built target flash
Run CPack packaging tool...
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: stlink
CPack: - Install project: stlink
CPack: Create package
CPack: - package: /home/usr/stlink.git/build/Release/stlink-1.4.0-39-g6db0fc2-amd64.deb generated.**

**~/stlink.git $ cd /home/usr/stlink.git/build/Release/
usr@usr-desktop ~/stlink.git/build/Release $ sudo dpkg -i *.deb
Selecting previously unselected package stlink.
(Reading database ... 311484 files and directories currently installed.)
Preparing to unpack stlink-1.4.0-39-g6db0fc2-amd64.deb ...
Unpacking stlink (1.4.0-39-g6db0fc2) ...
Setting up stlink (1.4.0-39-g6db0fc2) ...
Processing triggers for man-db (2.7.5-1) ...**

PirikitoOreo, 2018-06-24:
I was trying to upload test firmware to my stm32f103C8T6 using stlink-v2 and system workbench ide, but without success. I don't understand where is a problem. How can I check if my stlink-v2 is correctly installed?

xor-gate, 2018-06-24:
I would like to help you, but re-read what your are writing. Trying to understand what your problem is, i'm missing details (logfile, what did you type etc).

PirikitoOreo, 2018-06-24:
I would like to show you logfile, is it possible to find/generate it?

lyakh, 2018-07-01:
I had the same problem as the OP. What fixed it is running "make clean" before "make". Looks like some files got committed to git that shouldn't be there?

Nightwalker-87, 2020-02-21:
Closing #717 as useful feedback has been given. Please open a new ticket if necessary.

closed 2020-02-21 by Nightwalker-87