stlink-org/stlink #720: support STLink/v2 programmer firmware updates (via DFU)

nerilex, 2018-07-03:
I wrote a [tool to update the firmware of an STLink/v2 programmer][1] itself via DFU on base.

If there is interest I would bring the code into a state ready for a pull-request, if not I would change the code in other ways to make it a separate tool.

Currently my code still contains GPL-code (of myself) and lots of dirty hacks and debug stuff, which will be removed soon.

The update tool was tested with cheap STLink/v2-clones.

Best regards,
~  bg

[1]: https://c0de.pw/bg/stlink/tree/fw-update

xor-gate, 2018-07-03:
It would be nice to have it in texane/stlink, but it must be relicensed to the 3 Clause BSD to be merged. It is unacceptable for the project to mix GPL and permissive licenses.

nerilex, 2018-07-03:
relicensing won't be a problem. The GPL-stuff is only in an AES-implementation I copied from one of my other projects.

For now I have put the functionality into a separate tool called /st-fwupdate/.

xor-gate, 2018-07-07:
Feel free to wrap up and send a PR on github for the project so we can have a look for integration. Please also add some documentation under the doc/* folder, as markdown file and manpage. Thanks for you effort!

nerilex, 2019-01-19:
Could you create a feature branch for me, to work on the integration?

Since I somehow need feedback how to integrate the stuff, which extends the backend, matching the philosophy of the current devs :-)

xor-gate, 2019-02-09:
@nerilex sorry for the late reply I was fearly busy past weeks. You should fork the project, create a branch (e.g feature/issue-720) and add create a `[WIP]` in the title pull request for texane/stlink project. If something is not clear, feel free to ask. Thanks!

Nightwalker-87, 2020-04-09:
@nerilex: Sry for the delay. I just came across this issue during a review... Is your offer still open? :laughing: It certainly would be a "nice to have". Please let us know.

Nightwalker-87, 2020-06-03:
ping @nerilex

Nightwalker-87, 2021-12-04:
Closed due to inactivity and missing interest.

closed 2021-12-04 by Nightwalker-87