stlink-org/stlink #724: GD32F13

EmerickH, 2018-07-15:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [ ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ ] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

Output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:
`short description of the expected value`

Thank you,
The stlink project maintainers

xor-gate, 2018-07-15:
You need the reset pin, else it won't work in most situations.

EmerickH, 2018-07-15:
Ok... I'm going to retry with the reset and I will update the issue.

EmerickH, 2018-07-15:
I've soldered a cable on the board on the reset resistor and connected it to T_JRST
[image]
But now I have:
```
> st-flash --reset write build/hover.bin 0x8000000
st-flash 1.4.0-41-ge147a8e
2018-07-15T14:46:36 INFO usb.c: -- exit_dfu_mode
2018-07-15T14:46:36 INFO common.c: Loading device parameters....
2018-07-15T14:46:36 INFO common.c: Device connected is: F1 Medium-density device, id 0x13030410
2018-07-15T14:46:36 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2018-07-15T14:46:36 INFO common.c: Attempting to write 15404 (0x3c2c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08003c00 erased
2018-07-15T14:46:36 INFO common.c: Finished erasing 16 pages of 1024 (0x400) bytes
2018-07-15T14:46:36 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2018-07-15T14:46:36 ERROR flash_loader.c: unknown coreid, not sure what flash loader to use, aborting! coreid: ffffffff, chipid: 410
2018-07-15T14:46:36 WARN flash_loader.c: Failed to write flash loader to sram!
2018-07-15T14:46:36 ERROR common.c: stlink_flash_loader_init() == -1
stlink_fwrite_flash() == -1
```
(when I deconnect RST, I have the same log as previous)

xor-gate, 2018-07-15:
Thats weird, I have no clue why it doesn't work.

EmerickH, 2018-07-15:
Maybe somes clues here:
https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f1/flash.h
https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/f0/flash.h
https://github.com/libopencm3/libopencm3/blob/master/include/libopencm3/stm32/common/flash_common_f01.h
To recreate a flashloader for F1 like:
https://github.com/texane/stlink/blob/master/flashloaders/stm32f0.s
I will try tomorrow to continue digging...

EmerickH, 2018-07-30:
I never achieved to programm it...
(Maybe dead?)
I bought a new board with a STM32 and it is working well!
Should I close this issue?
Thank you

xor-gate, 2018-07-30:
Broken hardware can result in weird unexplainable errors. Thanks for reporting back it works now with good hardware!

Nightwalker-87, 2020-05-27:
Reopened this for investigation.

Nightwalker-87, 2020-07-18:
The GD32F130C8 is not officially supported by this toolset.
However it would indeed be worth to have a look this as a possible feature.
At first there should be further testing to find out more about current incompatibilities with this MCU series.

Nightwalker-87, 2021-05-10:
@Ant-ON: Can you review this in relation to the recently referenced commit?
It looks as if this has been resolved already, but I don't know since when exactly...

Ant-ON, 2021-05-11:
@Nightwalker-87 I think it most likely closes by #1044. The second problem with the flashloader launch ability is also likely to be closed by the flashloader improving (#1113, #1124).

Nightwalker-87, 2021-05-16:
Partially fixed in #1044, #1113 and finally closed with #1124.

closed 2021-05-16 by Nightwalker-87