stlink-org/stlink #734: Murata can't be programmed during sleep mode?

silardg, 2018-08-30:
- [ X] Programmer/board type: Stlink/v2-onboard
- [ X] Operating system: Linux=
- [X ] Target chip (and optional board):=cmwx1zzabz-091 murata

I have put my murata chip into sleep and can't get it to flash.
Output on boot0 tied to gnd (pulldown 10k)
[image]

Output on boot0 tied to vcc (pullup 10k)
[image]

I have tried:
- timing the reset button to get it to program: did not work
- boot0 to vcc
- different stlink
- different cable for stlink
- checked cables from stlink to board

Thanks!

xor-gate, 2018-09-01:
Hi @silardg, thanks for reporting. As far as I know stlink is not ready to support sleep mode debugging/programming. It requires some effort. In the meantime I point you to the more complete OpenOCD which should work.

silardg, 2018-09-03:
Thanks for the reply. The chip wasn't properly soldered on.

closed 2018-09-03 by silardg

Nightwalker-87, 2020-03-18:
Does not seem related to the st-link project anyway...