stlink-org/stlink #736: `st-util` can't find my Nucleo board

twilco, 2018-09-07:
- [x] Programmer/board type: Stlink/v2, Stlink/v2-onboard - One of these two I think?  It's a Nucleo board.
- [x] Programmer firmware version: Unknown; generic St-link module.
- [x] Operating system: Mac OS Sierra, version 10.12.6
- [x] Stlink tools version and/or git commit hash: Installed `stlink` v1.5.0 via `brew` (although st-util command output says version 1.4.0)
- [x] Stlink commandline tool name: `st-util`
- [x] Target chip (and optional board): e.g STM32L433RCT6P (NUCLEO-L433RC-P)

My Nucleo board is plugged into my MacBook Pro via USB.  There is one red light flashing, one green light flashing, and one green light lit and not flashing.  However, `st-util` doesn't appear to be able to find the device.  I was hoping to program the board using `openocd`, but it wasn't working which brought me here.  What is going wrong?

```
$ st-util        
st-util 1.4.0
2018-09-06T19:46:45 WARN usb.c: Couldn't find any ST-Link/V2 devices
```

```
$ openocd -f interface/stlink-v2-1.cfg -f target/stm32l4x.cfg
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : clock speed 480 kHz
Error: open failed
in procedure 'init' 
in procedure 'ocd_bouncer'
```

xor-gate, 2018-09-07:
That's indeed weird both complete different applications are unable to find the programmer on the nucleo. Normaly there should not be a problem. Maybe a permission problem (try running it with `sudo` just once)? Using OS X myself it works fine for other boards.

twilco, 2018-09-07:
No dice on using `sudo` - tried it on both commands with the same results.  Interestingly, it actually doesn't appear to show up in the list of USB devices, either.

https://imgur.com/q874WvK

This certainly seems like a problem :).  Is there anything special I need to do to make it recognize the Nucleo?  Admittedly, I haven't done anything besides install `stlink` and [the stuff found here](https://rust-embedded.github.io/discovery/03-setup/macos.html), so perhaps there are some setup instructions I'm missing.

xor-gate, 2018-09-07:
No its just plug and play. Maybe your programmer is dead on your nucleo board. Or does it work on a different usb port or computer?

xor-gate, 2018-09-07:
It is possible the USB cable is broken, USB port is not functioning well or the board is not okay.

twilco, 2018-09-13:
So after trying 2 MacBook Pros, a Windows desktop, all USB ports on the previously mentioned devices, a USB hub, two different cables, and ordering a new board that also didn't work, I ordered a brand new, third cable...

That worked with both of the boards 😭.  To future readers, check your cables!

closed 2018-09-13 by twilco