stlink-org/stlink #737: Has 1.5.0 been retagged?

fxcoudert, 2018-09-13:
The 1.5.0 tarball at `https://github.com/texane/stlink/archive/1.5.0.tar.gz` had a SHA-256 checksum of ce26b3db73f0bdf87ced78a2a90f3d515914f7d1211b1ca4b9acdf1882ca9d81 when it was released, but the checksum is now 4f0301edf9e904688ba31cf181653102f367c747e23ad356de1c430d1a654e2e

Has the tarball been affected? We need to confirm it was intentionally updated (ruling out malicious intent) in order to restore the software in Homebrew: https://github.com/Homebrew/homebrew-core/issues/31779

xor-gate, 2018-09-13:
Hi @fxcoudert, i'm not 100% sure it is some time ago. I see there are 27 commits since 1.5.0 so I will release a new 1.5.1 with the new features and tag it. So we can forget the 1.5.0 release then.

barjac, 2019-11-11:
In current clone of git, (11th Nov. 2019) 'git describe' reports 1.4.0-99-g3690de9
Something is wrong :(