stlink-org/stlink #742: MacBook pro 2016- STM3L4 can't find device

pdedman, 2018-10-21:
I have a MacBook pro 2016 (4 thunderbolts) running macOS High Sierra connected to a STM32L475 and st-util returns the following error: 

"WARN usb.c: Couldn't find any ST-Link/V2 devices"

The device is connected and receiving power. Has anyone tested on the most recent MacBook?

Thanks!

xor-gate, 2018-10-21:
Which programmer do you use, onboard or external stlink?

pdedman, 2018-10-22:
I am not sure. I followed the directions in: https://github.com/texane/stlink/issues/499 running st-util from my MacBook terminal. However, I tried connecting my STM32 to my raspberry pi and it won't even detect it.

pdedman, 2018-10-22:
Ah. Sorry. Tried a third USB cable and it works now.

pdedman, 2018-10-22:
Found this https://github.com/texane/stlink/issues/736 and tried a third cable like OP. Works fine now. Thank you.

xor-gate, 2018-10-22:
Those problems are always hard to find. Thanks for updating the issue. Good luck!

closed 2018-10-22 by xor-gate