stlink-org/stlink #746: troff man pages show wrong dash style for options

reox, 2018-11-19:
Some options in the man pages of `st-info` and partially `st-util` use a single dash instead of a double dash.
For example, st-info has the option "--probe" (shown in the examples section
correctly), but the man pages says "-probe".
When using "st-info -probe", the tool just exists with a status code of 0.
The man page of st-util contains in the description hints to the options
"-listen_port" and "-stlinkv1" which seems to be also replaced by the double
dash variants.

The options are correctly written in the markdown files, but it seems that pandoc does this automatically. I'm no expert at pandoc but it seems it is this package causing it: https://pandoc.org/MANUAL.html#typography

See also Debian bug report: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=914103

xor-gate, 2018-11-19:
Hi @reox,

I have looked into the generation and seems we did not as pandoc expectation. Could you have a look?

[image]

There is only a very small issue left where something is not bold.

Let me know.

Kind regards,
Jerry

reox, 2018-11-19:
@xor-gate yep, that looks good! thank you!

xor-gate, 2018-11-19:
Fixed in latest master now! Thanks for reporting.

PR #747 (2018-11-19): doc/man: Fix double dash issue (fixes #746)
Fixes #746
commit f87076fc (2018-11-19): doc/man: Fix double dash issue (fixes #746)
closed 2018-11-19 by xor-gate