stlink-org/stlink #749: Invalid flash type, unknown memory region on STM32F779I

maked0n, 2018-12-10:
- [X] Programmer/board type: Stlink/v2-onboard
- [X] Operating system: Linux 4.14.3
- [X] Stlink tools version and/or git commit hash: v1.5.0
- [X] Stlink commandline tool name: e.g st-info, st-flash
- [X] Target chip (and optional board): e.g STM32F779I-EVAL

When I try to flash my STM32, I get this error:
```
$ st-flash --reset write intflash.bin 0x8000000
st-flash 1.5.0
2018-12-11T00:52:25 INFO common.c: Loading device parameters....
2018-12-11T00:52:25 WARN common.c: Invalid flash type, please check device declaration
Unknown memory region
```

It seems, that st-info utility is unable to detect my device:
```
$ st-info --chipid
0x0000
```

lsusb output (maybe it would be helpful):
```
$ lsusb
... 
Bus 001 Device 024: ID 0483:374b STMicroelectronics ST-LINK/V2.1
...
```

maked0n, 2019-02-06:
@nit1995, it was a problem with power supply. After I fixed my power adapter, all is ok.

lucaspeixotot, 2019-04-08:
My problem is not related to power supply, someone knows how to solve this?

ngoc-emg, 2019-08-18:
I have a same error. Did you have any solution for it? @lucaspeixotot

skypodolsky, 2019-08-23:
The same problem for me.

TomHyhlik, 2019-12-09:
Also stucked with the same problem.

JasonShiao, 2020-01-03:
I have a similar problem in #107 .
I try to short the reset pin to GND and then this `Unknown memory region`  error message is shown too.
I'm using Mac so I am not able to use ST-link Utility for Windows.
So far the only solution works for me is:
1. Short the reset pin to GND at first
2. Immediately unplug the reset pin and enter st-flash command simultaneously.
The timing is critical and it may take several times to succeed.

susundberg, 2020-01-24:
Confirm what @JasonShiao said -- i had similar trouble with STM32F103  "blue pill" board and ST-LINK V2 -- both were (too) cheap. Just adding this comment for rest of ppl out there. 

Did several tries push RESET to GND - release the RESET around the time first characters appear to terminal from ST-LINK.

Also had several "unknown chip id! 0x5fa0004" tries -- i would say same root cause