stlink-org/stlink #750: Flash verification error at 128K mark on STM32L4C6 (NUCLEO-L$32KC)

dhylands, 2018-12-11:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [ST-LINK/V2-1 onboard ] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [STSW-LINK007 2.32.22] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [Linux - Ubuntu 18.04 LTS ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ a201d3e5bd4a52a6e2a7f89947688f580c2b3554 (latest master)] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [st-flash write firmware2.bin 0x08000000] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ STM32L432C6] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

A as-detailed description possible of the problem with debug output when available.

The firmware files I was using is available here:
https://www.dropbox.com/sh/nxgho1y9yj4bgfj/AACn_ZOkEN2DRAsuXyCRe9G-a?dl=0

They are almost identical versions of Micropython. Of note is that they're both larger than 128K.

Output:
Flashing onto the NUCLEO-L432KC with an erased flash (i.e. first time or after `st-flash erase`) works as expected:
```
$ st-flash write firmware1.bin 0x08000000
st-flash 1.4.0-54-ga201d3e
2018-12-11T12:52:28 INFO common.c: Loading device parameters....
2018-12-11T12:52:28 INFO common.c: Device connected is: L43x/L44x device, id 0x10016435
2018-12-11T12:52:28 INFO common.c: SRAM size: 0xc000 bytes (48 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2018-12-11T12:52:28 INFO common.c: Attempting to write 219420 (0x3591c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08035800 erasedEraseFlash - Page:0x12b Size:0x800 
2018-12-11T12:52:34 INFO common.c: Finished erasing 108 pages of 2048 (0x800) bytes
2018-12-11T12:52:34 INFO common.c: Starting Flash write for F2/F4/L4
2018-12-11T12:52:34 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 22812
2018-12-11T12:52:39 INFO common.c: Starting verification of write complete
2018-12-11T12:52:41 INFO common.c: Flash written and verified! jolly good!
```

If I then try to flash firmware2.bin, then I get this error:
```
660 >st-flash write firmware2.bin 0x08000000
st-flash 1.4.0-54-ga201d3e
2018-12-11T12:52:59 INFO common.c: Loading device parameters....
2018-12-11T12:52:59 INFO common.c: Device connected is: L43x/L44x device, id 0x10016435
2018-12-11T12:52:59 INFO common.c: SRAM size: 0xc000 bytes (48 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2018-12-11T12:52:59 INFO common.c: Attempting to write 219428 (0x35924) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08035800 erasedEraseFlash - Page:0x12b Size:0x800 
2018-12-11T12:53:05 INFO common.c: Finished erasing 108 pages of 2048 (0x800) bytes
2018-12-11T12:53:05 INFO common.c: Starting Flash write for F2/F4/L4
2018-12-11T12:53:05 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 22820
2018-12-11T12:53:08 INFO common.c: Starting verification of write complete
2018-12-11T12:53:10 ERROR common.c: Verification of flash failed at offset: 131072
stlink_fwrite_flash() == -1
```

If I execute:
```
st-flash erase
```
then the above flash of firmware2.bin works as expected.

Expected/description:
Flashing firmware2.bin immediately after flashing firmware1.bin should work properly (i.e. no verification error) without having to do an intervening `st-flash erase`

Thank you,
The stlink project maintainers

dhylands, 2018-12-12:
The error seems to be coming from the `calculate_L4_page`.

The datasheet for the L432 says "Up to 256 KB single bank Flash".
Unfortunately, the reference manual doesn't even mention the dual bank option in the FLASH_OPTR register. Bit 21 is marked as reserved.

FLASH_OPTR is being read as 0xfffff8aa which has bit 21 set. If I remove the bank code from calculate_L4_page then everything works as expected.

So I'm not sure how to determine single-banked versus dual-banked in a generic way.

xor-gate, 2018-12-12:
Flash memory banking is indeed always a bit nasty to get right. Weird it has the bit reserved in the datasheet. I'm not very familiar with the flash drivers. If you have a solution feel free to submit an PR.

dhylands, 2018-12-14:
Fixed by #751

closed 2018-12-14 by dhylands

xor-gate, 2018-12-15:
Thanks @dhylands. Maybe nice to know is if you put e.g "fixes #750" within your PR the issue is auto-closed (see https://help.github.com/articles/closing-issues-using-keywords/).