stlink-org/stlink #752: O_BINARY mode for open() on Windows is needed

JiriBilek, 2018-12-20:
Thanks for a good work on a quite useful application.

- [X] Programmer/board type: STLink/v2
- [X] Programmer firmware version: V2.J32.S7
- [X] Operating system: Windows 7, mingw build
- [X] Stlink tools version and/or git commit hash: v1.5.1/git-0a2b7a4
- [X] Stlink command line tool name: `st-flash`
- [X] Target chip (and optional board): STM32F030

Reading flash memory into a binary file results in bigger files than expected (and corrupted). Analyzing the output I found out that in the binary file the byte `0x0d = CR` is added before every `0x0a = LF`. I am convinced it is caused by missing O_BINARY flag when opening the output file for writing.
The problem is relevant only for Windows build.

Example of command:
`st-flash --format=binary --flash=0x8000 read f.bin 0x08000000 0x7000`
Should read exactly 0x7000 bytes but in my case reads 0x7050.

The line with missing O_BINARY is: https://github.com/texane/stlink/blob/0a2b7a4ff076b877155a76d5000be14a73bf524b/src/common.c#L1367

Adding the O_BINARY flag and the bug is fixed. Do you want from me to make a PR?

xor-gate, 2018-12-21:
Hi @JiriBilek, please provide a PR. Thanks!

JiriBilek, 2018-12-21:
Closed via #753

closed 2020-03-20 by Nightwalker-87