stlink-org/stlink #754: Here's how to add support for STM32L412

jcw, 2018-12-27:
I've made the following changes to stlink to get it to recognise and upload to the L412:

```
$ git diff
diff --git a/src/chipid.c b/src/chipid.c
index 6f93af1..ca6e815 100644
--- a/src/chipid.c
+++ b/src/chipid.c
@@ -507,6 +507,19 @@ static const struct stlink_chipid_params devices[] = {
             .bootrom_base = 0x1ff00000,
             .bootrom_size = 0x2000
         },
+        {
+            // STM32L41X
+            .chip_id = STLINK_CHIPID_STM32_L41X,
+            .description = "L412 device",
+            .flash_type = STLINK_FLASH_TYPE_L4,
+            .flash_size_reg = 0x1fff75e0,    // "Flash size data register" (sec 49.2, page 1809)
+            .flash_pagesize = 0x800,         // Page erase (2 Kbyte) (sec 3.2, page 93)
+            // SRAM1 is 32k at 0x20000000
+            // SRAM2 is 8k at 0x20008000
+            .sram_size = 0x0A000,
+            .bootrom_base = 0x1fff0000,
+            .bootrom_size = 0x7000
+        },
         {
             // unknown
             .chip_id = STLINK_CHIPID_UNKNOWN,
diff --git a/src/flash_loader.c b/src/flash_loader.c
index 7684680..f2e6367 100644
--- a/src/flash_loader.c
+++ b/src/flash_loader.c
@@ -306,6 +306,7 @@ int stlink_flash_loader_write_to_sram(stlink_t *sl, stm32_addr_t* addr, size_t*
         loader_code = loader_code_stm32f0;
         loader_size = sizeof(loader_code_stm32f0);
     } else if ((sl->chip_id == STLINK_CHIPID_STM32_L4) ||
+              (sl->chip_id == STLINK_CHIPID_STM32_L41X) ||
               (sl->chip_id == STLINK_CHIPID_STM32_L43X) ||
               (sl->chip_id == STLINK_CHIPID_STM32_L46X) ||
               (sl->chip_id == STLINK_CHIPID_STM32_L4RX) ||
```

I'm not sure about `bootrom_size`, but the rest should be correct.

Note also that `STLINK_CHIPID_STM32_L41X` (0x464) was already present in `src/chipid.h`.

xor-gate, 2018-12-28:
Please provide a PR (https://help.github.com/articles/about-pull-requests/)

jcw, 2019-01-24:
I hope you won't take offense, but no, sorry, I can't be bothered. I was not asking for a favour, merely offering a suggestion. Feel free to close this issue.

xor-gate, 2019-01-24:
I like the change, but you get credits forever when you provide a mergeable PR. So I don't spend time on it :+D

jcw, 2019-01-24:
easy: 1) make the change, 2) mention `closes #754` in the commit msg, 3) done

EliKrumholz, 2019-03-27:
I implemented a similar change without seeing this first and ended up with all the same values as @jcw, and it appears to be working fine with the NUCLEO-L412KB board. @xor-gate Can I open up a pull request (https://github.com/EliKrumholz/stlink/commit/348149b6d41139253d289f4380166d491e3fec2c), or were there any other issues that needed to be addressed?

lukebayes, 2020-02-10:
I'm struggling to get st-link to work on an STM32L412x chip and stumbled over this thread. I got excited and thought maybe this change was never integrated, but alas, it was.

For future readers, the change suggested by @jcw above was merged in May of 2019.

Thanks!

Nightwalker-87, 2020-02-21:
Please verify if the problem still exists in Release v1.6.0.

lukebayes, 2020-02-21:
I can confirm, that I was able to get STLINK to upload directly to two different STM32L412xx chips using both FTDI and SWD without any changes to software.

Nightwalker-87, 2020-02-21:
Fixed since commit 189f253ce3f9a6a39d0ebf7b2b5c23defddd757c which is part of Release v1.6.0.

closed 2020-02-21 by Nightwalker-87