stlink-org/stlink #755: [Linux Mint] `send_recv send request failed: LIBUSB_ERROR_TIMEOUT`

geri0102, 2018-12-29:
I have a problem flashing my STM32F407VET6 with an ST-LINK V2. Below is my terminal output (I'm running as root with `su`). I have also tried Wireshark on the USB: [wireshark_usbmon3_20181229002230.pcapng.gz](https://github.com/texane/stlink/files/2715921/wireshark_usbmon3_20181229002230.pcapng.gz)

```
gergo-desktop Letöltések # st-flash write mbed_blinky.ARCH_MAX.bin 0x8000000
st-flash 1.4.0-54-ga201d3e
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_RESETSYS
2018-12-29T00:05:36 INFO common.c: Loading device parameters....
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
2018-12-29T00:05:39 ERROR common.c: Failed to read core_id
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
2018-12-29T00:05:42 WARN common.c: unknown chip id! 0x4b4e494c
gergo-desktop Letöltések # st-info --probe
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
 serial: 303030303030303030303031
openocd: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
gergo-desktop Letöltések # lsusb
Bus 011 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 010 Device 002: ID 1c4f:0034 SiGma Micro 
Bus 010 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 003 Device 002: ID 0e8f:00a4 GreenAsia Inc. 
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 003: ID 1a2c:2d23 China Resource Semico Co., Ltd 
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
gergo-desktop Letöltések # st-info --probe
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
 serial: 303030303030303030303031
openocd: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
gergo-desktop Letöltések # st-flash --version
v1.4.0-54-ga201d3e
gergo-desktop Letöltések # uname -rvs
Linux 4.15.0-29-generic #31~16.04.1-Ubuntu SMP Wed Jul 18 08:54:04 UTC 2018 
```

xor-gate, 2019-01-03:
Thats weird, could you try to update the firmware of the programmer ?

xor-gate, 2019-01-03:
Also check for wonky USB cables and correct SWD connection

geri0102, 2019-01-04:
It was a brand new cable and st-link. I tried to disconnect and reconnect the SWD connection. I'll try to update the firmware as soon as I arrive home. Also I'll try another st-link if I can get one quick.

geri0102, 2019-01-04:
You were right about firmware (it was V2J16S4).But even after that i got errors and the st-info --probe shoved 
```
Found 1 stlink programmers
serial: 31006b063142323851190843
openocd: "\x31\x00\x6b\x06\x31\x42\x32\x38\x51\x19\x08\x43"
flash: 0 (pagesize: 0)
sram: 0
chipid: 0x0000
descr: unknown device
```
I started swapping around cables... It did nothing. Then I disassembled the ST-Link, and found out the problem: a bad solder joint. So I soldered it, now it finally works. Thanks for your help. @ @xor-gate

xor-gate, 2019-01-05:
That's awesome! If it works could you close the issue?

closed 2019-01-05 by geri0102