stlink-org/stlink #759: Feature request: support for G0 chips

WRansohoff, 2019-01-13:
It'd be nice to support the fairly new STM32G0 line of chips.

- [X] Programmer/board type: Stlink/v2
- [X] Programmer firmware version: V2.J29.S7
- [X] Operating system: Ubuntu 18.04 (all platforms?)
- [X] Stlink tools version and/or git commit hash: git-7651d2116fd74c7803ea00ab1da7cf3d00faf44c
- [X] Stlink commandline tool name: st-util, st-flash
- [X] Target chip (and optional board): STM32G071GB

Right now they aren't supported:

```
2019-01-12T16:04:23 INFO common.c: Loading device parameters....
2019-01-12T16:04:23 WARN common.c: unknown chip id! 0x20006460
```

And it's possible that I'm reading the manuals wrong, but it also looks like the core flash register offsets (CR, SR, KEYR) are slightly different from most other STM32 lines.

I wrote a small patch that seems to work for erasing and flashing an STM32G071GB chip, but I have only tested it with a small LED-blinking example which only occupies a single page of Flash:

https://github.com/WRansohoff/stlink/commit/c1ad2cf6f7bed2c22f18dfdd0087cd0d5e8b9997

And besides the lack of testing, I don't know how well that conforms to the rest of the project's code style and formatting, so it's probably not worth making a pull request yet.

xor-gate, 2019-01-13:
Hi @WRansohoff, thanks for improving stlink. I had a look at your code and it looks fine. You may provide a PR!

WRansohoff, 2019-01-13:
Sure, here - I do want to clean it up a bit, but it's something to start with:

https://github.com/texane/stlink/pull/760

yaqwsx, 2019-04-25:
I am trying to use `st-flash` with STM32G071GB. It works well for small programs, however, once the program size is larger than 4096 bytes, it fails to verify the program (`--debug` log here: https://pastebin.com/5dtTYqp9): 

```
st-flash 1.5.1-24-gfeb8661
2019-04-25T15:49:58 INFO common.c: Loading device parameters....
2019-04-25T15:49:58 INFO common.c: Device connected is: G071/G081 device, id 0x20006460
2019-04-25T15:49:58 INFO common.c: SRAM size: 0x9000 bytes (36 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 2048 bytes
2019-04-25T15:49:58 INFO common.c: Attempting to write 5040 (0x13b0) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08001000 erased
2019-04-25T15:49:58 INFO common.c: Finished erasing 3 pages of 2048 (0x800) bytes
Writing
Starting   2 page write
2019-04-25T15:49:58 INFO common.c: Starting verification of write complete
2019-04-25T15:49:59 ERROR common.c: Verification of flash failed at offset: 4096
stlink_fwrite_flash() == -1
```
Is it possible there is a bug in the patch introduced by @WRansohoff?

closed 2020-03-20 by Nightwalker-87