stlink-org/stlink #764: st-flash and st-link require sudo in ubuntu

renn0xtek9, 2019-01-21:
On Ubuntu and deirvated distros, st-flash and st-util require sudo to work properly.
Adding them to sudoers (via visudo) remove the need for password but not for sudo.
Adding the user to dialout group won't change anything.
Adding a GROUP="myusername",\      line to the udev rules won't change anything.
It would be very much appreciated to provide documentation on how to remove this need for sudo, so that it will make integration with IDEs (like VSCode and others) very much easier.

kmfarley11, 2019-01-24:
How did you install stlink?
If you do the following you should not need sudo to run the commands (so long as it is not run as root):
```
git clone https://github.com/texane/stlink.git
cd stlink/build
cmake ..
make && sudo make install
```
Note that you may also need to update your udev rules if you have not already. 
[stlink_udev_rule.tar.bz2](http://www.openstm32.org/dl160)
`sudo tar -xvf stlink_udev_rule.tar.bz2 -C /etc/udev/rules.d`

dhylands, 2019-01-25:
It works fine for me under ubuntu 18.04.

To investigate, start by plugging your STLINK device into the usb port. Then run lsusb. You should see an entry something like the following:
```
Bus 005 Device 017: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)
```
Note the bus number (005) and the Device (017). You should then do:
`ls -l /dev/bus/usb/005/017`    (replacing 005 and 017 appropriately).

On my system I see the following:
```
crw-rw-rw- 1 root root 189, 528 Jan 24 17:52 /dev/bus/usb/005/017
```
which is world writable (this is from the MODE:="0666" below). I have several files in my /etc/udev/rules.d directory. In this particular case, the 49-stlinkv2-1.rules file contains the following:
```
# stm32 nucleo boards, with onboard st/linkv2-1
# ie, STM32F0, STM32F4.
# STM32VL has st/linkv1, which is quite different

SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="374b", \
    MODE:="0666", \
    SYMLINK+="stlinkv2-1_%n"

# If you share your linux system with other users, or just don't like the
# idea of write permission for everybody, you can replace MODE:="0666" with
# OWNER:="yourusername" to create the device owned by you, or with
# GROUP:="somegroupname" and mange access using standard unix groups.
```
and the idVendor od 0483 and idProduct of 374b matches the vendor id from the lsusb output.

Make sure that you have all 3 files from here: https://github.com/texane/stlink/tree/master/etc/udev/rules.d in your /etc/udev/rules.d directory. After copying new files or editing excisting files in /etc/udev/ruled.d you should run the following:
```
sudo udevadm control --reload-rules
sudo udevadm trigger
```

to ensure that the rules actually take effect. Using the trigger command means that you shouldn't need to unplug and replug the device, but you might want to try that for good measure as well.

If the VID:PID of your device doesn't match those in any of the 3 files, then you may need to create a custom rule file to match your VID:PID.

renn0xtek9, 2019-01-25:
Hello and thanks for you answer 
I did all this 
`ls -l /dev/bus/usb/001/011 
crw-rw-rw-+ 1 root root 189, 10 Jan 25 19:56 /dev/bus/usb/001/011
`
It is still the same. 
By adding the command in sudo visudo, i don't have to enter password anymore but still need sudo.
That is weird

xor-gate, 2019-02-02:
@dhylands would you like to update the documentation with the extensive description? At https://github.com/texane/stlink/blob/master/doc/compiling.md#permissions-with-udev.

Nightwalker-87, 2020-03-25:
@renn0xtek9: What is the current state of this? I believe you use v1.5.1, is that correct? Can you help us with some more investigation?
@dhylands: May I copy-paste your description to our documentation?

davehylands, 2020-03-25:
@Nightwalker-87 By all means - feel free.

Nightwalker-87, 2020-04-02:
@davehylands: Thank You for your helpful contribution and feedback.

Closing this with commit f7ab09f1148d6930df25bb8bf591a6e0e1e68137 as we know by now that the udev configuration is functional.
I can confirm this with a recent debian system as well.

The problem appears to be an isolated issue on a local installation that we can not reproduce.
A concluding advice is to retry with a clean (system) installation with default user permissions.

commit f7ab09f1 (2020-04-02): General Project Update

- [doc] Updated list of supported devices
- [doc] verify udev-rules (Closes #764)
- Removed old issue-templates.
- Updated BSD-3 license file.
commit d3c11fbe (2020-04-02): General Project Update

- [doc] Updated list of supported devices
- [doc] verify udev-rules (Closes #764)
- Removed old issue-templates.
- Updated BSD-3 license file.
closed 2020-04-02 by Nightwalker-87