stlink-org/stlink #766: STM32F401 Nucleo-64 no longer programmable

Quadricopter, 2019-01-28:
Since commit 7651d2116fd74c7803ea00ab1da7cf3d00faf44c, comment following line makes st-flash work again with Nucleo-64
https://github.com/texane/stlink/blob/30de1b30e7a96974ae764d0508c151eadb1addd1/src/flash_loader.c#L265

lsusb:
Bus 001 Device 024: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)

xor-gate, 2019-02-09:
We should continue in #761 because this is duplicate. We should skim through the PRs which broke the flashing.

Nightwalker-87, 2020-03-20:
Fixed in commits f5d0454ab0a18f209ad5b7d0d51f3945b27dd892 and b40c2436b68a35544a1eae10f4b6a3de7648a566

Nightwalker-87, 2020-03-20:
Fixed in commits f5d0454ab0a18f209ad5b7d0d51f3945b27dd892 and b40c2436b68a35544a1eae10f4b6a3de7648a566.

closed 2020-03-20 by Nightwalker-87

Quadricopter, 2020-03-24:
Hello,

tested with master branch ( commit: f5d0454ab0a18f209ad5b7d0d51f3945b27dd892 )
**It's OK for me!**

> st-flash write nucleo.bin 0x8000000
st-flash 1.6.0-31-gf5d0454
2020-03-24T19:46:31 INFO common.c: Loading device parameters....
2020-03-24T19:46:31 INFO common.c: Device connected is: F4 device (Dynamic Efficency), id 0x10016433
2020-03-24T19:46:31 INFO common.c: SRAM size: 0x18000 bytes (96 KiB), Flash: 0x80000 bytes (512 KiB) in pages of 16384 bytes
2020-03-24T19:46:31 INFO common.c: Attempting to write 12368 (0x3050) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08000000 erased
2020-03-24T19:46:32 INFO common.c: Finished erasing 1 pages of 16384 (0x4000) bytes
2020-03-24T19:46:32 INFO common.c: Starting Flash write for F2/F4/L4
2020-03-24T19:46:32 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 12368
2020-03-24T19:46:32 INFO common.c: Starting verification of write complete
2020-03-24T19:46:32 INFO common.c: Flash written and verified! jolly good!