stlink-org/stlink #772: Script error when building on GNU/Linux

vanfanel, 2019-02-19:
- [X] Operating system: Linux
- [X] Stlink tools version and/or git commit hash:  git-1165cf7

Hi there,

I'm trying to build on GNU/Linux (Debian-based distro, Lubuntu) but I get an error when trying. All dependencies are met. Yet there seems to be a script problem:

```
manuel@vanfanel:~/src/stlink$ make release
-- The C compiler identification is GNU 8.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- stlink version: 1165cf7
--           Major  Minor  Patch 
-- Performing Test C_SUPPORTS_STD_GNU99
-- Performing Test C_SUPPORTS_STD_GNU99 - Success
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Performing Test C_SUPPORTS_O2
-- Performing Test C_SUPPORTS_O2 - Success
-- Performing Test C_SUPPORTS_WERROR
-- Performing Test C_SUPPORTS_WERROR - Success
-- Found Libusb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- found USB
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gtk+-3.0'
--   Found gtk+-3.0, version 3.24.1
-- Looking for sys/mman.h
-- Looking for sys/mman.h - found
-- Looking for unistd.h
-- Looking for unistd.h - found
CMake Error at CMakeLists.txt:123 (set_target_properties):
  set_target_properties called with incorrect number of arguments.

-- Manpage generation disabled
-- Debian-based Linux OS detected
-- Configuring incomplete, errors occurred!
See also "/home/manuel/src/stlink/build/Release/CMakeFiles/CMakeOutput.log".
See also "/home/manuel/src/stlink/build/Release/CMakeFiles/CMakeError.log".
make: *** [Makefile:43: build/Release] Error 1

```

In CmakeLists.txt there's this:

```
set_target_properties(${STLINK_LIB_SHARED}
        PROPERTIES SOVERSION ${PROJECT_VERSION_MAJOR}
        VERSION ${STLINK_SHARED_VERSION}
)

```

So, any idea on what's going on here?

Thanks!

tuffstudio, 2019-05-03:
I second that, same issue on MBP Mojave 10.14.4

alf, 2019-05-05:
I can confirm the issue on Mojave. I cheated and succeeded in compiling by hard coding `${PROJECT_VERSION_MAJOR}` to `1` in CMakeLists.txt:124:

```
- 	PROPERTIES SOVERSION ${PROJECT_VERSION_MAJOR}
+ 	PROPERTIES SOVERSION 1

xor-gate, 2019-05-06:
yeah the versioning is utterly broken at this point 🤕

flux242, 2019-06-26:
@xor-gate put at least a short remark into the CMakeList near the line 124 on how to solve the problem until it's properly fixed. Because the error is anything but intuitive to solve

Nightwalker-87, 2020-03-01:
@slyshykO @bmarvo: What are your suggestions on this? Unfortunately I was not able to reproduce this on my debian-based buildsystem. Maybe someone else has more luck with this...

slyshykO, 2020-03-01:
Steps to reproduce a similar error :

```sh
git clone https://github.com/texane/stlink.git --no-tags
cd ./stlink
make release
```

slyshykO, 2020-03-21:
Hello @vanfanel. Can you check the develop branch if this issue is gone?

Nightwalker-87, 2020-03-25:
@tuffstudio @alf @flux242: Can one of you verify this?

Nightwalker-87, 2020-03-28:
@slyshykO: **vanfanel**, **tuffstudio**, **alf** and **flux242** all have no fork of this project anymore... (so I don't expect any reply here)
I personally don't have any build problems with Debian Bullseye (testing).
However I can't reproduce this issue from commit 1165cf712713af48c61c4782ad5c9477c4574708 either. Maybe the commit reference is wrong in the opening post. I'll try an earlier release (e.g. 1.5.1 and 1.6.0), as I feel like I have seen similar before somewhere, and let you know.

Nightwalker-87, 2020-03-28:
I can reproduce the problem in v1.5.1, but not in v1.6.0 (working). So in fact it was already fixed in the latter release. I conclude that your PR #872 was helpful, but did not fix this, as it had been fixed before. Never mind, I'll add the relevant PR from before and then close this as resolved & verified.

Nightwalker-87, 2020-03-28:
This compilation error was fixed in #762. 
Additional further improvements were made with #872.

closed 2020-03-28 by Nightwalker-87

Nightwalker-87, 2020-04-07:
Duplicate of #563.

commit 677a3bef (2020-03-21): Merge pull request #872 from slyshykO/detect-version

make Version.cmake more error-resistant (Fixes #772)
commit 4d55b6b3 (2020-05-31): make Version.cmake more error-resistant. fix #772