stlink-org/stlink #774: How do we control NRST pin via GDB?

ceremcem, 2019-03-04:
I'm successfully using st-link for STM32F103 and STM32F407 devices. I noticed no need for NRST pin to be able to fully manage a debug session. [`reload`](https://github.com/aktos-io/mcu-debug/blob/3a45c0972969850e968c604ee4d46a0e4a3e87aa/gdb-init#L30-L61)ing code just works (including restarting the MCU). 

However, when I use STM32F030 device, it turns out it's necessary to generate a NRST signal in order to properly restart the MCU. As an example, I created a blinker example which generates a square wave on a pin and load the code to the MCU. Pin does not alter its state. When I manually connect the NRST pin to the GND and release, pin starts generating square wave as intended. 

I connected a logic analyzer to the NRST pin of the debugger hardware. Neither loading the code directly (`st-flash write`) nor loading via GDB generates NRST pulse. However, `st-flash erase` generates this NRST pulse. 

# Question 

Question comes to the point: How do we generate `NRST` pulse via gdb command in order to properly restart the MCU?

xor-gate, 2019-03-09:
This is a very good question, I'm not sure when the jtag reset function is called from GDB. The lowlevel programmer command is implemented here https://github.com/texane/stlink/blob/30de1b30e7a96974ae764d0508c151eadb1addd1/src/usb.c#L395-L414. For the GDB server in st-util it is located here https://github.com/texane/stlink/blob/b9c315d990abfde3008a917e767c63d2c1c1ddf2/src/gdbserver/gdb-server.c#L1234-L1250. So probably sending a `jtag_reset` command through the gdb `monitor` should work (interactive shell `monitor jtag_reset`).

ceremcem, 2019-03-09:
`monitor jtag_reset` works as expected. Thanks.

closed 2020-03-20 by Nightwalker-87