stlink-org/stlink #777: ldconfig: /usr/lib/libstlink.so.1 is not a symbolic link

spacerace, 2019-03-14:
After installating st-link util into my system I see this error on every run of apt-get and when I run ldconfig manually. My system is debian stretch (stable).
```
....
Setting up cups-bsd (2.2.1-8+deb9u3) ...
Setting up libkprintutils4 (4:4.14.26-2) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
ldconfig: /usr/lib/libstlink.so.1 is not a symbolic link

Processing triggers for man-db (2.7.6.1-2) ...
Processing triggers for hicolor-icon-theme (0.15-1) ...
Setting up libokularcore7 (4:16.08.2-1+deb9u1) ...
Setting up libkexiv2-11 (4:15.04.3-1) ...
ldconfig: /usr/lib/libstlink.so.1 is not a symbolic link

Setting up okular (4:16.08.2-1+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
ldconfig: /usr/lib/libstlink.so.1 is not a symbolic link

# 
```

I have a look at the specified file(s) in /usr/lib:
```
# ls -al /usr/lib/libstlink*
-rw-r--r-- 1 root root 463280 Mar 10 11:45 /usr/lib/libstlink.a
-rwxr-xr-x 1 root root 287816 Mar 10 11:45 /usr/lib/libstlink.so
-rwxr-xr-x 1 root root 287816 Mar 10 11:45 /usr/lib/libstlink.so.1
-rwxr-xr-x 1 root root 287816 Mar 10 11:45 /usr/lib/libstlink.so.1.5.1
```

Now my workaround (or maybe the solution?) is this:
```
# cd /usr/lib/
/usr/lib# rm libstlink.so.1
/usr/lib# ln -s libstlink.so.1.5.1 libstlink.so.1
/usr/lib# ls -al libstlink*
-rw-r--r-- 1 root root 463280 Mar 10 11:45 libstlink.a
-rwxr-xr-x 1 root root 287816 Mar 10 11:45 libstlink.so
lrwxrwxrwx 1 root root     18 Mar 14 11:41 libstlink.so.1 -> libstlink.so.1.5.1
-rwxr-xr-x 1 root root 287816 Mar 10 11:45 libstlink.so.1.5.1
/usr/lib# ldconfig
/usr/lib#
```

After this my system's ldconfig isn't complaining anymore about non-symlink.
I also replaced libstlink.so with a symlink to libstlink.so.1.5.1, which was not neccessary for getting rid of the ldconfig-message, but saves ~280k of disk space.

I don' acutally have a question about this, as it seems, that I have found a solution, but maybe some could explain this to me/us and maybe this could be fixed by changing the install routine in your Makefiles.

But there is another question. Wouldn't it be better to install st-link into `/opt/st-link-revision/` and extend user's PATH env. variable as st-link is no "systemwide-default-unix-tool".
My arm-gcc is also located in `/opt/gcc-arm-none-eabi-8-2018-q4-major`, I could install st-link utility into that toolchain also...
What is the best place to install?

- [X] Operating system: linux, debian stretch (stable).
- [X] Stlink tools version and/or git commit hash: from 2 days ago (How do I get git revision?)

Nightwalker-87, 2020-03-21:
I believe this is not something common users should need to think about...

Nightwalker-87, 2020-03-21:
@spacerace: Please do a clean build with v1.5.1 first so that you can ensure that there are no leftovers and then retry. If not successful you may move on to v1.6.0. Additional feedback to us would be helpful indeed.

Nightwalker-87, 2020-04-02:
I can't reproduce this error with debian bullseye (testing) on v1.5.1.
The installation completes without any warnings or errors.

I finally find this in `/usr/lib/` (which also is how it should look like):

```
-rw-r--r-- 1 user user 127808 Apr  3 00:45 libstlink.a
lrwxrwxrwx 1 user user     14 Apr  3 00:46 libstlink.so -> libstlink.so.1
lrwxrwxrwx 1 user user     18 Apr  3 00:46 libstlink.so.1 -> libstlink.so.1.5.1
-rw-r--r-- 1 user user  91920 Apr  3 00:45 libstlink.so.1.5.1
```
From this one can conclude that your installation paths are somehow messed up.
A helpful advice would be to `make clean` and then remove all remnants in the destination directories. Rerun the installation procedure afterwards in order to achieve a clean setup.

closed 2020-04-02 by Nightwalker-87

Nightwalker-87, 2020-04-02:
My advice for installing the toolchain is to install it from the package repository of the distribution. The predefined installation routine then ensures that the toolchain is installed to the right directory paths and all symbolic links and system variables are set correctly, thus the toolchain is properly integrated into the system.