stlink-org/stlink #782: LIBUSBX_API_VERSION undef

goog, 2019-03-18:
```bash
build on ubuntu12.04
```
cheng@ubuntu:~/stlink$ pkg-config --libs --cflags libusb-1.0
-I/usr/include/libusb-1.0  -lusb-1.0 

/home/cheng/stlink/src/sg.c: In function ‘stlink_open’:
/home/cheng/stlink/src/sg.c:969:5: error: "LIBUSBX_API_VERSION" is not defined [-Werror=undef]
cc1: all warnings being treated as errors

xor-gate, 2019-03-18:
This is bug, it must compile on Ubuntu. I know there are some differences between libusb versions.

goog, 2019-03-27:
what i have changed will effect flash program?  @xor-gate

xor-gate, 2019-03-31:
Sorry @goog for the late reply, no it will not effect the working of the flashing of the binary data to the controller.

xor-gate, 2019-05-31:
We need a patch for this

Nightwalker-87, 2020-02-22:
Please verify if the problem still exists in Release v1.6.0.
I can find these code lines in /src/sg.c, as this problem seems to be related to #733 which targetied FreeBSD. Therefore this does not help you with Ubuntu: `#if defined (__FreeBSD__)`
This condition should be defined in such a way that all affected operating systems are listed and not only FreeBSD. Please comment on this.

slyshykO, 2020-02-22:
LIBUSB_API_VERSION defined since libusb 1.0.13.
The lastest supported Ubuntu 16.04 provide libusb 1.0.20.

Also LIBUSB_API_VERSION  for FreeBSD almost 4 years (https://svnweb.freebsd.org/base/head/lib/libusb/libusb.h?revision=301957&view=markup)

Nightwalker-87, 2020-02-22:
Thanks for the feedback @slyshykO !

Debian Jessie (oldoldstable), which is currently the oldest Debian distribution version supported, uses libusb 1.0.19. Taking the Debian project as a reference for longterm support (oldoldstable branch) this would mean that we are looking back to 2015 when this version was released. This is 5 years in time. Looking at this, I feel like no one should expect the stlink project to ensure full software compatibility dating back any longer. We don't do ourself any favour with this.

To me this would imply that we completely throw out `LIBUSBX_API_VERSION` and have it replaced with `LIBUSB_API_VERSION`. @bmarvo any other thoughts on this?

Nightwalker-87, 2020-03-23:
@slyshykO: #211 told me something about the background of this issue. This is where `LIBUSBX_API_VERSION` comes from. I think we should get rid of this now, as it is still found in `/src/sg.c` L940ff. It smells... <º))))>< ...rotten. :-D We neither need it nor do I like to keep it.

slyshykO, 2020-03-23:
@Nightwalker-87 
This is short summary https://github.com/libusb/libusb/wiki/FAQ#libusborg_libusbxorg_and_libusbinfo .

Nightwalker-87, 2020-03-23:
Yes, I came across that somewhere else in relation to the mentioned issue above.
It doesn't change my opinion though...

Nightwalker-87, 2020-03-23:
EDIT: **...BUT** as #733 reveals by the state of August 2018: At least FreeBSD v11.3 still seems to rely on it, as they indeed appear to use an old version. Can a FreeBSD user clarify or confirm this to ensure we do not run into a regression here? BTW: What about Fedora (see #211)? We may not care about the remaining distros in that list.

slyshykO, 2020-03-23:
> 
> 
> LIBUSB_API_VERSION defined since libusb 1.0.13.
> The lastest supported Ubuntu 16.04 provide libusb 1.0.20.
> 
> Also LIBUSB_API_VERSION for FreeBSD almost 4 years (https://svnweb.freebsd.org/base/head/lib/libusb/libusb.h?revision=301957&view=markup)

This

Nightwalker-87, 2020-03-23:
Sorry, I missed the second part of it. :$
Let's get rid of it then... <º))))>< --> \_/

slyshykO, 2020-03-23:
My opinion is since libusbx reunited with libusb in **2014** we should not care about it.

Nightwalker-87, 2020-03-23:
I'm on about this: 

> ... is still found in `/src/sg.c` L940ff.

Nightwalker-87, 2020-03-23:
I agree that cleaning this up does close this ticket as well as #211.

Nightwalker-87, 2020-03-23:
Commit d2b5f8dbfbb0458e90cde4c90de025e7ae518b7b removes support for `LIBUSBX_API` (deprecated since 2016) and closes this issue.

Nightwalker-87, 2020-03-26:
Reopening this, as topic is not solved by now.

closed 2020-04-02 by Nightwalker-87