stlink-org/stlink #783: Uninstall/reinstall from source on mac

hierophect, 2019-03-18:
Hello, 

My setup is on Mac 10.14.2, STlink2, targeting Bluepill. 

I'd like to make modifications to the project source in order to support the full 128k of memory available on the STM32F103C8T6, used on the Bluepill boards, to match the official ST tool and allow me to use platforms like mbed. However, I'm not sure from the project documentation how to perform an uninstall/reinstall from source since it recommends using Homebrew. 

I've got the normal st-link program installed, along with all the dependencies like the arm-none-eabi-gc, etc. Is there a simple way that I can uninstall the main program and reinstall the edited version, without touching the dependencies, and without installing the official version off homebrew? 

Sorry for the newbie question, I do usually tend to use homebrew for everything, so I'm not that familiar with source installations on mac!

hierophect, 2020-03-18:
This does not need to be re-opened, the compile from source option for mac is well documented in the [compiling manual](https://github.com/texane/stlink/blob/master/doc/compiling.md). 

The specific fix I was recommending for the bluepill support is in the **src/common.c** file, in the stlink_load_device_params() function, in the following text:

```
} else if (sl->chip_id == STLINK_CHIPID_STM32_L1_CAT2) {
    sl->flash_size = (flash_size & 0xff) * 1024;
} else if ((sl->chip_id & 0xFFF) == STLINK_CHIPID_STM32_L1_HIGH) {
    // 0 is 384k and 1 is 256k
    if ( flash_size == 0 ) {
        sl->flash_size = 384 * 1024;
    } else {
        sl->flash_size = 256 * 1024;
    }
} else {
    sl->flash_size = flash_size * 1024;
}
``` 

replacing the final else with 

```
} else {
     sl->flash_size = flash_size * 1024;
     /* HACK for STM21F103C8T6 */
     if (sl->flash_size == 64 * 1024 && getenv("C8T6HACK"))
       sl->flash_size = 128 * 1024;
}
```

This is soundly a hack however, supporting an undocumented chip feature (albeit one that is a leading source of popularity for the blue pill board).

Nightwalker-87, 2020-03-18:
Well that may be of general interest though as these boards are quite common. So as a maintainer of this project I'd consider this useful and would keep this ticket open for one of the upcoming releases. However this does not imply that you need to keep up with this issue.

hierophect, 2020-03-19:
@Nightwalker-87 sorry, I only meant to imply that I no longer needed any help on it! If you support adding special cases like the Bluepill that's great.

Nightwalker-87, 2020-03-19:
I understand. So I will close this as resolved then.

Note on Bluepill-Support:
For 128k support also take a look at issue 575.
As far as I am aware of, the tools currently support the STM32F103C8T6 as an original device.
Problems seem to occur with fake chips marked as STM32F103C8T6, but turn out to be CKS32F103C8T6 instead as revealed by their chip ID. See issues 756 and 833.
I suggest you open a new issue and PR with your 128k proposal. We should not recycle this issue here with a different topic than the original one.

Nightwalker-87, 2020-03-26:
ping @hierophect

closed 2020-03-26 by Nightwalker-87

hierophect, 2020-03-26:
I'll make a new issue on this today

Nightwalker-87, 2020-03-26:
You may add a comment to #516, which already covers this.