stlink-org/stlink #784: Odd solution to Couldn't find any ST-Link/V2 devices

fd9750, 2019-03-19:
Hi,
I tried to run **st-flash write ./build/blink.bin 0x08000000** from within VSCODE using a Mingw32 Shell to call the command on my Windows machine.
Initially it did not work and I got the error message as described in the title. I did a bit of digging and I found another issue here ( #651 )where it was suggested to use a USB2.0 port instead of a USB3.0 port and a mention later to update the USB host controller.

I tried the first suggestion and sure enough it worked. I then tried to update all possible USB host controller but they all reported being 100% up to date.

I then plugged the ST-link device back into the USB3.0 and ever since then it works perfectly, regardless of which type of USB port I use.

I am not complaining but I am really puzzled how come it now works, any suggestion would be much appreciated.

pan-mroku, 2019-03-21:
My guess: I don't remember the details exactly but Windows saves some info about which drivers it used for which usb port. After some time it gathers so many different drivers that it gets lost. I observe this with memory card readers and once in a while I have to use USBDeview (but there are other programs) in order to delete all this history. Otherwise it won't write to the card. Changing port does help for some time. Maybe trying to update host controller did the same thing?

Nightwalker-87, 2020-03-21:
@fd9750 @pan-mroku: What is the current state of this? Have you found a solution you can tell us about?