stlink-org/stlink #785: Unknown chip id 0x10006463

CharlieA0, 2019-03-25:
**NOTICE: The issue may be closed without notice when not enough information is provided!**

Thank you for giving feedback to the stlink project. Take some time to fill out
 check boxes with a X in the following items so developers and other people can try to
 to find out what is going on. And add/remove what is appropriate to your problem.

When submitting a feature request, try to reuse the list and add/remove what is appropriate.
 Place a `X` between the brackets `[X]` to mark the list item.

- [X] Programmer/board type: Stlink/v2-onboard
- [ ] Programmer firmware version: V2J28M18
- [X] Operating system: Ubuntu 18.04
- [X] Stlink tools version and/or git commit hash:  v1.3.0
- [X] Stlink commandline tool name: `st-flash` and `st-flash`
- [X] Target chip (and optional board): e.g STM32F413ZHT (32F413HDISCOVERY)

I'm trying to flash a STM32F4 Discovery but I've suddenly started getting the following when running `st-flash write main.bin 0x8000000`

Output:
```
2019-03-24T20:10:32 INFO /opt/stlink-1.3.0/src/common.c: Loading device parameters ...
2019-03-24T20:10:32 WARN /opt/stlink-1.3.0/src/common.c: unknown chip id! 0x10006463
```

Expected/description:

It wasn't the confirmation of a successful flash that I expected. I also tested with another board of the same type and got the same result. I then tried flashing with openocd and got an error "Cannot identify target as STM32 family" with the same incorrect device id on both boards. I was finally successful flashing with the ST-Link Utility on Windows 10.

I would appreciate any advice. Thank you.

xor-gate, 2019-04-03:
It should work https://github.com/texane/stlink/blob/master/include/stlink/chipid.h#L72, but for some unknown reason it doesn't

iceseyes, 2019-06-02:
Same problem with my custom board. When I use v1.5.0 o v1.5.1 board was halted as I run any command (st-info --probe, st-flash, st-util) and then no mcu was seen. If a keep power on button pressed on while I flash it works. 
Using v1.4.0 board remains working and I can flash many times board without using power on botton.

closed 2020-03-31 by Nightwalker-87