stlink-org/stlink #787: flash error

goog, 2019-03-29:
cheng@ubuntu:~/stlink/build/Release$ ./st-flash write /home/cheng/stm32f4/arm-rtems4.11/c/stm32f4/testsuites/samples/hello/hello.bin 0x8000000
st-flash 1.5.1-21-gc6836b4
2019-03-29T00:43:07 INFO usb.c: -- exit_dfu_mode
2019-03-29T00:43:07 INFO common.c: Loading device parameters....
2019-03-29T00:43:07 INFO common.c: Device connected is: F4 device, id 0x10076413
2019-03-29T00:43:07 INFO common.c: SRAM size: 0x30000 bytes (192 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 16384 bytes
2019-03-29T00:43:07 INFO common.c: Attempting to write 180852 (0x2c274) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08020000 erasedEraseFlash - Sector:0x5 Size:0x20000 
2019-03-29T00:43:08 INFO common.c: Finished erasing 6 pages of 131072 (0x20000) bytes
2019-03-29T00:43:08 INFO common.c: Starting Flash write for F2/F4/L4
2019-03-29T00:43:08 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
2019-03-29T00:43:47 ERROR flash_loader.c: flash loader run error
2019-03-29T00:43:47 ERROR common.c: stlink_flash_loader_run(0x8000000) failed! == -1
stlink_fwrite_flash() == -1

xor-gate, 2019-04-03:
Please use the issue template which you fully removed when creating the ticket, this is not enough information. See https://raw.githubusercontent.com/texane/stlink/master/.github/ISSUE_TEMPLATE.md

Nightwalker-87, 2020-02-21:
Closed due to lack of provided information.

Nightwalker-87, 2020-03-19:
@goog: Please verify if this problem persists with Release v1.6.0.

closed 2020-06-17 by Nightwalker-87