stlink-org/stlink #788: Build errors

sharma-kullu, 2019-03-31:
```
[RELEASE] 
Scanning dependencies of target stlink 
[  3%] Building C object CMakeFiles/stlink.dir/src/chipid.c.obj
[  6%] Building C object CMakeFiles/stlink.dir/src/common.c.obj /home/alam/stlink/src/common.c:19:0: error: "O_BINARY" redefined [-Werror]  
#define O_BINARY 0 //! @todo get rid of this OH MY (@xor-gate)  
In file included from /usr/include/sys/fcntl.h:3:0,                  
from /usr/include/fcntl.h:12,                  
from /home/alam/stlink/src/common.c:10: 
/usr/include/sys/_default_fcntl.h:75:0: note: this is the location of the previous definition 
#define O_BINARY _FBINARY  cc1: all warnings being treated as errors make[3]: *** [CMakeFiles/stlink.dir/build.make:87: 
CMakeFiles/stlink.dir/src/common.c.obj] Error 1 
make[2]: *** [CMakeFiles/Makefile2:68: CMakeFiles/stlink.dir/all] Error 2 
make[1]: *** [Makefile:150: all] Error 2 
make: *** [Makefile:28: release] Error 2
```
OS: Cygwin for windows (64 bit)

xor-gate, 2019-04-03:
This is problem because under Cygwin O_BINARY is defined and under Mingw it is not. So there should be a extra compiler definition check to the custom memmap *NIX libc C function.

Nightwalker-87, 2020-03-20:
@sharma-kullu: Any update on this?

Nightwalker-87, 2020-04-07:
Closed by commit bf840a1ae82ffc3ef9929f243bb8050d6856f698.

closed 2020-04-07 by Nightwalker-87

Nightwalker-87, 2020-04-08:
Closed by commit 9f2cad2c978a55a4125e6b8c4c656a59e7f9448c.

commit bf840a1a (2020-04-07): Avoid re-define of O_BINARY on Windows

(Fixes #788)
commit 74a665d6 (2020-04-08): Avoid re-define of O_BINARY on Windows

(Fixes #788)
commit 9f2cad2c (2020-04-08): Avoid re-define of O_BINARY on Windows

(Fixes #788)