stlink-org/stlink #791: unknown chip id

goog, 2019-04-08:
ubuntu12.04 64bit STM32F407ZGT6

A as-detailed description possible of the problem with debug output when available.

Output:

```
cheng@ubuntu:~/stlink/build/Release$ ./st-flash write /home/cheng/stm32f4/arm-rtems4.11/c/stm32f4/testsuites/samples/hello/hello.bin 0x8000000
st-flash 1.5.1-21-gc6836b4
2019-04-07T23:22:21 INFO common.c: Loading device parameters....
2019-04-07T23:22:21 WARN common.c: unknown chip id! 0x5fa0004
```

set321go, 2019-05-26:
I'm seeing the same issue when trying to program STM32F042K6, which was talking to `st-info --chipid` initially. I do notice that that `0x5fa0004` is the reset command.

Nightwalker-87, 2020-03-19:
@goog: Please verify if the problem still exists in Release v1.6.0.

Nightwalker-87, 2020-04-02:
Resolved by either #715 (swapped SWDIO and SWCLK lines) or #832 (firmware).

closed 2020-04-02 by Nightwalker-87