stlink-org/stlink #792: STM32F746 fail to program

maidenone, 2019-04-11:
- [ STLINK/V2] Programmer/board type: e.g Stlink/v1, Stlink/v2, Stlink/v2-onboard
- [2.J33.S7 ] Programmer firmware version: e.g STSW-LINK007 2.27.15
- [ Linux ] Operating system: e.g Linux, Mac OS X, Windows (with specific version)
- [ v1.5.1-23-g224f6ee ] Stlink tools version and/or git commit hash: e.g v1.1.0/git-c722056
- [ st-flash ] Stlink commandline tool name: e.g `st-info`, `st-flash`, `st-util`
- [ STM32F746NG] Target chip (and optional board): e.g STM32F402VG (STM32Fxxx Discovery)

Description:
When i try to flash a .bin file, only 0x8000000 to 0x8001060 gets written and verification fails. 

Output:

```
sudo st-flash write file.bin 0x08000000
st-flash 1.5.1-23-g224f6ee
2019-04-11T10:06:43 INFO common.c: Loading device parameters....
2019-04-11T10:06:43 INFO common.c: Device connected is: F7 device, id 0x10016449
2019-04-11T10:06:43 INFO common.c: SRAM size: 0x50000 bytes (320 KiB), Flash: 0x100000 bytes (1024 KiB) in pages of 2048 bytes
2019-04-11T10:06:43 INFO common.c: Attempting to write 492772 (0x784e4) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08040000 erasedEraseFlash - Sector:0x5 Size:0x40000 
2019-04-11T10:06:48 INFO common.c: Finished erasing 6 pages of 262144 (0x40000) bytes
2019-04-11T10:06:48 INFO common.c: Starting Flash write for F2/F4/L4
2019-04-11T10:06:48 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 1252
2019-04-11T10:06:59 INFO common.c: Starting verification of write complete
2019-04-11T10:06:59 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1

```

from: sudo st-flash --debug write file.bin 0x08000000 > debug.log 2>&1

```
> st-flash 1.5.1-23-g224f6ee
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_GET_CURRENT_MODE
> 2019-04-11T10:35:48 DEBUG common.c: stlink mode: unknown!
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_GET_CURRENT_MODE
> 2019-04-11T10:35:51 DEBUG common.c: stlink mode: unknown!
> 2019-04-11T10:35:51 DEBUG common.c: *** stlink_enter_swd_mode ***
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_DEBUG_ENTER
> 2019-04-11T10:35:54 DEBUG common.c: *** looking up stlink version
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_GET_VERSION
> 2019-04-11T10:35:57 DEBUG common.c: *** set_swdclk ***
> 2019-04-11T10:35:57 DEBUG common.c: *** stlink_reset ***
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_DEBUG_RESETSYS
> 2019-04-11T10:36:00 INFO common.c: Loading device parameters....
> 2019-04-11T10:36:00 DEBUG common.c: *** stlink_core_id ***
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_DEBUG_READCOREID
> 2019-04-11T10:36:03 ERROR common.c: Failed to read core_id
> [!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
> [!] send_recv STLINK_JTAG_READDEBUG_32BIT
> 2019-04-11T10:36:06 WARN common.c: unknown chip id! 0x19a0010
> 2019-04-11T10:36:06 DEBUG common.c: *** stlink_close ***
```

Expected/description:
`succeed flashing/verification`

Side note:
Works fine with stlink utility in windows, not a connection / hw fault.
Have not managed to get it to work in openocd, get the error:

```
Error: timed out while waiting for target halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0xffffffd8
Error: error waiting for target flash write algorithm
```

AndKe, 2019-10-09:
Have you found a solution ?
I have my stlink v2  (10-pin, from ebay) doing this: 
```
$ st-info --probe
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_RESETSYS
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
Found 1 stlink programmers
 serial: 303030303030303030303031
openocd: "\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x30\x31"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device

```

Nightwalker-87, 2020-03-19:
@maidenone: Please verify if the problem still exists in Release v1.6.0.

AndKe, 2020-03-19:
it works fine for me. my error was that I had two "alike" StLink2's with a little different pinout.

Nightwalker-87, 2020-03-19:
Ah ok. Are you talking about STLink-v2 _clones_ with different pinouts?

AndKe, 2020-03-19:
yes, I think that is what was my problem, in any case, I can program them now.

Nightwalker-87, 2020-03-19:
Ok, thx for the feedback. Closing this issue now as resolved.

closed 2020-03-19 by Nightwalker-87