stlink-org/stlink #793: st-info probe failed  Nucleo 144-H743zi

wnear, 2019-04-13:
Operating System: MacOS 10.14.4
Programmer/board type: Stlink/v2-onboard
Target chip: STM32F743ZI Nucleo-144

output:

```
st-info --probe                                            
Found 1 stlink programmers
serial: 303636444646353435323537373137                                                           
openocd: "\x30\x36\x36\x44\x46\x46\x35\x34\x35\x32\x35\x37\x37\x31\x37" 
flash: 0 (pagesize: 0)                                                                          
sram: 0                                                                                        
chipid: 0x0000
descr: unknown device
```

I've install the stlink-sever, updated with stsw-link007.  The st official programming software, STM32CubeProgrammer works all right. 

I've tried the brew install version, source code make version, and the source code stm32h7xx-improvements branch version, all not working. 

It worked before I upgrade my OS.

xor-gate, 2019-04-14:
It should work, it seems stlink doesn't detect the programmer correctly as the serial string is not correct. And no chipid is recognised. It could also be a problem with libusb dependency on latest 10.14 macOS.

xor-gate, 2019-04-14:
I have tested on my STM32F767 Nucleo-144 devboard and latest master seems to work for me:
```
bash-3.2$ uname -a
Darwin Jerrys-iMac.local 18.5.0 Darwin Kernel Version 18.5.0: Mon Mar 11 20:40:32 PDT 2019; root:xnu-4903.251.3~3/RELEASE_X86_64 x86_64
bash-3.2$ ./build/Release/st-info --probe
Found 1 stlink programmers
 serial: 303636444646343835353530373535
openocd: "\x30\x36\x36\x44\x46\x46\x34\x38\x35\x35\x35\x30\x37\x35\x35"
  flash: 2097152 (pagesize: 2048)
   sram: 524288
 chipid: 0x0451
  descr: F76xxx device
```

wnear, 2019-04-16:
> I have tested on my STM32F767 Nucleo-144 devboard and latest master seems to work for me:
>
> ```
> bash-3.2$ uname -a
> Darwin Jerrys-iMac.local 18.5.0 Darwin Kernel Version 18.5.0: Mon Mar 11 20:40:32 [...]

The "serial"  can only be 303636444646343835353530373535?
I uninstall and install libusb ,libusb-compat, stlink again, still not working. 
I check the system report and get the following , does this matter:

STM32 STLink:

  Product ID:	0x374b
  Vendor ID:	0x0483  (STMicroelectronics)
  Version:	1.00
  Serial Number:	066DFF545257717867114632
  Speed:	Up to 12 Mb/sec
  Manufacturer:	STMicroelectronics
  Location ID:	0x14210000 / 9
  Current Available (mA):	500
  Current Required (mA):	300
  Extra Operating Current (mA):	0
  Media:
microcontroller:
  Capacity:	2.1 MB (2,134,016 bytes)
  Removable Media:	Yes
  BSD Name:	disk3
  Logical Unit:	0
  Partition Map Type:	Unknown
  USB Interface:	1

lschuermann, 2019-05-26:
I've got the exact same board and am affected by this issue. However, the `serial` and `openocd` string differs which is expected, right?

I don't have any experience with STLink or STM32 chips, but I'd love to help if there's anything I can do to get this fixed. Running `v1.5.1`.

fpistm, 2019-05-26:
It seems there is a misunderstanding on this issue. OP talks about H7 not F7.
I guss, This is linked to #671.

xor-gate, 2019-05-27:
Close in favor of #671

closed 2020-10-21 by Nightwalker-87