stlink-org/stlink #794: [GUI] Missing function full erase

rtek1000, 2019-04-21:
Hi,

How to perform full erasure?

Full erase is useful for working with bootloader. I have tried bootloader type DFU / HID, where the VID / PID changes after uploading the sketck. So if the bootloader is burned on the chip not completely cleaned, the bootloader does not operate completely.

Thank you.

[image]

rtek1000, 2019-04-21:
Hi,

I was able to perform the deletion via the command line, using 'st-flash erase' (After the build. Note: It would be cool if it had a precompiled binary too, because Manjaro Linux does not have cmake preinstalled).

And I noticed that after the command 'st-info --probe' the LED indicator is in the 'connected' state (red). The GUI is blue when you click disconnect.

So how to do for st-info disconnect from the stlink programmer?

Thank you.

rtek1000, 2019-04-21:
Hi,

It would be nice for the GUI to display a status information after executing commands as it does not show anything after burn the bootloader.

Thank you.

Nightwalker-87, 2020-03-25:
@rtek1000: I agree, we are missing this feature in the GUI. It does belong there indeed.

Nightwalker-87, 2020-04-14:
I've been playing around with this a bit, but clearly need some advice, as I'm not very familiar with GTK. However it looks as if I could somehow manage to prepare the required graphical gtk elements in the code. They are without a function linked to it yet though.

Not really sure whether the current implementation strictly follows a Model View View-Modell (MVVM) scheme. However this should be considered, if one wants this to be implemented well. Also that would allow good maintenance. Could someone else have a look at this and assist?

Nightwalker-87, 2020-04-17:
@cedricboudinet: What do you think about this? Any help would be appreciated.

Nightwalker-87, 2021-12-04:
Closed due to inactivity and missing interest.

closed 2021-12-04 by Nightwalker-87