stlink-org/stlink #795: Windows binary installation instructions?

Andy2No, 2019-04-24:
Given, particularly, the problem raised in #738, it seems there is probably only one correct way to install the current binary for Windows (v1.3.0).

However, as far as I can see, there are no instructions or guidelines on how to do that.

Please explain the steps required, and document them somewhere they will be seen, when trying to install it.

At the moment, it just says this:

[b]Installation

Windows users can download v1.3.0 from the releases page.[/b]

- but the binary contains no further information on what to do to install it, and neither does the downloads page.

xor-gate, 2019-05-06:
It has been a very long time when I build the windows release, there are some people on the stm32duino forum which have newer build http://stm32duino.com/viewtopic.php?t=2515

fpistm, 2019-05-06:
Even this build is old 2017 😉

xor-gate, 2019-05-06:
Yes, you could get your hands dirty and build one yourself https://github.com/texane/stlink/blob/master/doc/compiling.md#windows-mingw64 :=D

Andy2No, 2019-05-06:
Thanks.  I'll look into that :)

Nightwalker-87, 2020-02-23:
Any update on this?

Andy2No, 2020-02-23:
Any idea how to get past this error?

"CMAKE_MAKE_PROGRAM is not set"

I have followed the Windows installation instructions (I have some notes to add to that), and I can run cmake from the command line, but I haven't yet got any further than this:

C:\Users\andy\stlink\scripts>mingw64-build.bat
CMake Error: CMake was unable to find a build program corresponding to "MinGW Ma
kefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a differe
nt build tool.
CMake Error: CMake was unable to find a build program corresponding to "MinGW Ma
kefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a differe
nt build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
'mingw32-make' is not recognized as an internal or external command,
operable program or batch file.
'mingw32-make' is not recognized as an internal or external command,
operable program or batch file.
'mingw32-make' is not recognized as an internal or external command,
operable program or batch file.

slyshykO, 2020-02-23:
@Andy2No , Do you have installed mingw64? If so, you should add it to the PATH.

Andy2No, 2020-02-23:
I had installed mingw64, and chose the option to add shortcuts to the Start menu.

As is often the case in Windows 7, I then couldn't find it in the Start menu.  Cmake was still highlighted, but that wasn't.

It seems they expect us to use a shortcut there, to launch a terminal with the environment already configured.  It was in

Start Menu -> Programs -> MinGW-W64 project -> i686-8.1.0-posix-dwarf-rt_v6-rev0

and the shortcut is called "Run terminal"

That starts at the root of C, so a CD is needed to get to the working directory:

C:\>cd C:\Users\andy\stlink\scripts

C:\Users\andy\stlink\scripts>mingw64-build.bat

- now appears to have worked.  I'll find a way to try it out, later.  I've been using the ST GUI based one, so far.

I now have st-info.exe and st-flash.exe, both in the scripts folder, where I ran the build script... I wasn't really expecting them to be there, so maybe that should be added to the installation instructions.... maybe I've just missed that part.

Other suggested changes - I had a bit of trouble finding what to download for MinGW64.  The link given in the installation instructions lead to me downloading the source.  I found the .exe here:

http://mingw-w64.org/doku.php/download

(or Downloads, on http://mingw-w64.org/doku.php)

I think giving that link instead, would make the instructions clearer.

Also, I haven't used git clone a lot, so I had to look that up.  A lot of people won't have used it at all, but it seems to be the best way to get sources from a git repository (especially if there are dependencies on other repositories), so a little more information would be useful there.

This worked (from my User folder, in a command prompt):

git clone https://github.com/texane/stlink

but someone using git from the command line for the first time would also need to do:

git init

If you feel like linking to a reference, I found it at:
https://git-scm.com/book/en/v2/Git-Basics-Getting-a-Git-Repository
2.1 Git Basics - Getting a Git Repository
(if not used before, read the part about git init)

For future reference, I've installed the current version of Cmake, with cmake-3.17.0-rc1-win64-x64.msi, and the current version of MingGW64, which is 8.1.0, using mingw-w64-install.exe

Andy2No, 2020-02-23:
... which installed the MingGW64 binaries here:

C:\Program Files (x86)\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0

- but I don't seem to need to know that, provided I use their "Run terminal" shortcut.

Andy2No, 2020-02-23:
I've attached the zipped package that was created, and the contents of the terminal window, mostly containing the compiler output.... subject to testing, when I get back up to speed on what I was intending to use it for, originally.

[build log.txt](https://github.com/texane/stlink/files/4241350/build.log.txt)
[stlink-1.6.0-3-g3af8048-win32.zip](https://github.com/texane/stlink/files/4241351/stlink-1.6.0-3-g3af8048-win32.zip)

Nightwalker-87, 2020-02-23:
I think @slyshykO was referring to the $PATH system variable within the Windows system settings, not to the directory path / install path.

Andy2No, 2020-02-23:
> 
> 
> I think @slyshykO was referring to the $PATH system variable within the Windows system settings, not to the directory path / install path.

Yes, I assumed that was what he meant.  There's also a path in the mingw64-build.bat file, and I missed that part of the instructions, but neither seems to matter, if you use the terminal shortcut provided with MinGW64 8.1.0 - I used the mingw64-build.bat file as supplied, without edits.

Windows does a slightly mysterious thing with the environment, so you don't necessarily need to alter it in system settings.  A batch file like the one they provided (via the shortcut) is able to achieve the same thing, by setting environment variables first...

It does this:

echo off
set PATH=C:\Program Files (x86)\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0\mingw32\bin;%PATH%
rem echo %PATH%
rem cd "C:\Program Files (x86)\mingw-w64\i686-8.1.0-posix-dwarf-rt_v6-rev0\mingw32\bin"
cd "C:\"
"C:\Windows\system32\cmd.exe"

Glaeqen, 2020-03-08:
Maybe I overlooked it in this thread but building `st-link` suite using MSVC toolchain for Windows 10 (x86-64) is suprisingly easy and straightforward.
1. Install MS Visual Studio Build Tools (includes msbuild, compiler, linker etc. - whole VS IDE is unnecessary)
2. Install CMake for Windows
3. Make sure everything is in your PATH (`msbuild` and `cmake`)
    - Tip: `msbuild` in default installation presumably should be in `C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\MSBuild\Current\Bin` directory
4. In repo directory: `mkdir build; cd build; cmake ..; msbuild <SLN_FILE>.sln /property:Configuration=Release`
5. Copy MS64 variant of libusb to a release directory containing executables (When DLL is not present, running execs causes prompt just to return with no error whatsoever. Annoying as hell)
6. st-util.exe is built somewhere in src/ directory; it has to be copied over.
7. Make sure you have [USB drivers](https://www.st.com/en/development-tools/stsw-link009.html) for ST-Link installed.

Done.

There's a [st-utils.zip](https://github.com/texane/stlink/files/4303224/st-utils.zip) ZIP file containing all built executables.

Andy2No, 2020-03-10:
Like most things, it's easy when you know how, Glaeqen.

I've detailed my process with decoding the instructions.  Maybe that will help someone else, who is similarly confused by them.

Glaeqen, 2020-03-11:
> 
> Like most things, it's easy when you know how, Glaeqen.
> 

Sorry, I did not mean to brag or to offend anyone. By _suprisingly easy and straightforward_ I was mostly referring to how flawlessly whole process went through - in contrary to my expectations. In my experience using msbuild is rather painful (I have more linux background) and this was a pleasant surprise.

I'm annoyed as well as others that there are no ready to use executables. Especially, when toolchain supports platform just fine.

Andy2No, 2020-03-11:
No offence taken :)

I'm pleased to see this thread has still been left open.  There's a tendency on github to close threads, which effectively hides them away from most people who might have wanted the same information.

Nightwalker-87, 2020-03-11:
@Andy2No: This ticket will be closed as soon as the related documentation has seen improvements that address this issue. This is currently planned for the upcoming release. If there are any further specific suggestions, please feel free to submit a commit.
@Glaeqen: I like your suggestion/proposal to use MSVS. To me this sounds like a suitable replacement for the (as of today) somehow outdated MinGW-Tools.
@bmarvo: What's your opinion on this?

Nightwalker-87, 2020-03-20:
Is anybody willing to go through a windows binary installation and would write some documentation to complement our README.md?

Nightwalker-87, 2020-04-07:
@Glaeqen: Could you please open a PR to add the necessary support to build on MSVS?

Nightwalker-87, 2020-04-07:
@Andy2No: I have updated the README (23c071edea45f6e8852fef52d884a680973d6d8f), which is not the final state though, I know, but at least represents the current situation. Additionally our compiling instructions have been rewritten (27aa88821197d3ffe82baff4e971c3488ec39899).

Windows binaries are now built into a separate build folder during installation, but that routine still wants some clean-up. I'm working on it...

Glaeqen, 2020-04-20:
@Nightwalker-87 I'm sorry I left you with this issue for such a long time. I created PR #942 as you wished. I fought yesterday with develop branch, it appears you fixed it overnight. Nice job, CMake scripts now look nice and clean :)

Nightwalker-87, 2020-04-21:
@Glaeqen: Thx for your help will look at it as soon as I find time. Still working on CMake packaging and will need some more time for that. Right now a working development branch has the highest priority. :sweat:

commit 06c66ed3 (2020-05-10): Updated documentation

- Added note on windows binary installation
- Contributors now listed in separate file

(Closes #738) (Closes #795)
PR #955 (2020-05-19): Update for package distribution
Changes & additions:

- New cpack package-config for DEB and RPM
- Toolchain file for cross-building
- Script to automate MinGW cross-building
- Added file headers for .cmake files
- Added dpkg and rpm to required pkgs
- Removed deprecated debian pkg config settings
- Update for travis build configuration:
--> Added builds for clang -m32 on arch: AMD64
--> Corrected invalid config settings
--> Added clang-9 builds for linux
--> Moved scripts to root directory
--> Added MinGW cross-test-build for linux
- Updated steps for release preparation
- Changed destination [... cut]
closed 2020-05-19 by Nightwalker-87