stlink-org/stlink #800: Library documentation

BigfootN, 2019-05-07:
Hello,

I would like to know where I can find (apart from reading the headers), if possible, any documentation on how to use the `stlink` library ?

Thank you very much in advance.

xor-gate, 2019-05-07:
Thats a good question there is a very very minimal example here https://github.com/texane/stlink/tree/6db0fc243ffb90aed314ef1d59811486e9214883/doc/app-example. The library is not (very well) documented, the library is used for lowlevel access on the stlink programmer and is not often used outside the project.

Nightwalker-87, 2020-03-25:
@BigfootN: Did this help you any further? I agree that we need more documentation on this.

Nightwalker-87, 2021-05-16:
@Ant-ON You recently mentioned that you use the library itself. What would you document on how to use it from your personal experience?

Nightwalker-87, 2022-08-27:
Closed due to inactivity and missing interest.

closed 2022-08-27 by Nightwalker-87