stlink-org/stlink #806: stty: /dev/ttyUSB0: No such file or directory

sevgicorumlu, 2019-05-24:
sudo ./test_stm32f0.sh 

stty: /dev/ttyUSB0: No such file or directory
./test_stm32f0.sh: 6: ./test_stm32f0.sh: st-flash: not found
./test_stm32f0.sh: 7: ./test_stm32f0.sh: cannot open /dev/ttyUSB0: No such file
0a1
> c8a71edd37b496ed9f1c763b86f1614b24215280e6d4c48b6cdf477f9f92af6c (what it means)

Nightwalker-87, 2020-03-19:
@sevgicorumlu: Can you shortly describe what action solved this issue for you? Thx in advance.

Nightwalker-87, 2020-04-02:
It appears that `st-flash` is either not installed or the binary-path is missing in the $PATH system variable. No USB device is detected at address `/dev/ttyUSB0`. As the content of the custom script remains unknown, no further help can be provided.

closed 2020-04-02 by Nightwalker-87