stlink-org/stlink #807: flashing error on F303K8

sawik-PL, 2019-05-25:
- [X] Programmer/board type: Nucleo-F303K8 Stlink/v2-onboard
- [X] Operating system: Linux
- [X] Stlink tools version and/or git commit hash: 1.5.1-28-gd040db5
- [X] Stlink commandline tool name: `st-util -m -v`
- [X] Target chip (and optional board): e.g STM32F303K8 Nucleo-F303K8

From debug output:
2019-05-25T22:01:13 ERROR flash_loader.c: write error, count == 1023
2019-05-25T22:01:13 ERROR common.c: stlink_flash_loader_run(0x8003800) failed! =
when trying to flash app image (from eclipse or gdb) - please note, that app begins not at the beginning of the flash region (offset is 0x3000)

Log attached
[issue-log.txt](https://github.com/texane/stlink/files/3219701/issue-log.txt)

xor-gate, 2019-05-28:
Are you sure there is no flash write protection?

sawik-PL, 2019-05-28:
Just checked to be 100% sure. The sequence is following
> $ st-flash erase
st-flash 1.5.1-28-gd040db5
2019-05-28T21:04:02 INFO common.c: Loading device parameters....
2019-05-28T21:04:02 INFO common.c: Device connected is: F3xx medium density device, id 0x10016438
2019-05-28T21:04:02 INFO common.c: SRAM size: 0x3000 bytes (12 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
Mass erasing

Then flashing bootloader:
> st-flash write build/STR-bootloader.bin 0x8000000
st-flash 1.5.1-28-gd040db5
2019-05-28T21:06:47 INFO common.c: Loading device parameters....
2019-05-28T21:06:47 INFO common.c: Device connected is: F3xx medium density device, id 0x10016438
2019-05-28T21:06:47 INFO common.c: SRAM size: 0x3000 bytes (12 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 2048 bytes
2019-05-28T21:06:47 INFO common.c: Attempting to write 9660 (0x25bc) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08002000 erased
2019-05-28T21:06:47 INFO common.c: Finished erasing 5 pages of 2048 (0x800) bytes
2019-05-28T21:06:47 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2019-05-28T21:06:47 INFO flash_loader.c: Successfully loaded flash loader in sram
  5/5 pages written
2019-05-28T21:06:48 INFO common.c: Starting verification of write complete
2019-05-28T21:06:48 INFO common.c: Flash written and verified! jolly good!

Bootloader works
Then running
> st-util -m -v

and trying to load app from eclipse/gdb
- works once
- doesn't work next times
- but works via command line "st-flash write build/STR-app.bin 0x8003000"

Nightwalker-87, 2020-07-05:
This wants verification with Release v1.6.0 and at least comparable hardware.

Nightwalker-87, 2021-03-25:
What is the actual definition of `stlink_flash_loader_run(0x8003800)`?

Ant-ON, 2021-04-06:
@Nightwalker-87 I think this should be fixed by #1113
ps 0x8003800 is not an error code, it's the write address.

Nightwalker-87, 2021-04-06:
Closed by #1113.

closed 2021-04-06 by Nightwalker-87